grblHAL/core #1: A Logo proposal

mnesarco, 2021-02-23:
Hi Friends,
I want to propose a new Logo:

[image]

If you like it I can send the svg.

terjeio, 2021-02-25:
@phil-barrett has made one as well. Used [here](https://www.grbl.org/projects). His is more geared towards use on breakout boards than beeing the one used for the firmware? So IMO having two distinct logos sharing the same basic style would be nice.

Perhaps a good idea to move this to a discussion in the original repository so more users can participate? I guess it will be a little while before this repository get much attention...

mnesarco, 2021-02-25:
@terjeio Ok, no problem. I put it here because I thought you had planned to move everything here. I have a much more refined version, I will open an Issue in the other repo.

terjeio, 2021-02-25:
> I put it here because I thought you had planned to move everything here.

I plan to do that eventually, but I want to see this repo gaining some traction first. There are a number of followers and watchers that I believe get notified about activity in the other repo.

phil-barrett, 2021-02-25:
I have a number of grblHAL treatments.  Not married to any of them, just needed something to use. I'm getting some odd effects when I click on the logos in the post preview.  Not sure what that is about.
[image]
[image]
[image]

mnesarco, 2021-02-26:
Discussion moved to: https://github.com/terjeio/grblHAL/discussions/268

mnesarco, 2023-07-17:
...2 years later discussion moved here again :D

Anyone interested in discussing a logo proposal?

[image]

Or should this issue be closed?

phil-barrett, 2023-07-17:
It seems ok to me.  To be honest, in the ensuing period, grblHAL has clearly inherited the Grbl crown. So, tying it back to "Classic" Grbl is not such a necessity anymore.  Grbl_ESP32, the only real contender, was renamed to FluidNC, breaking from the Grbl identity. Not sure what they were trying to do - IMO, it makes it another bit of obscure CNC software. There is a Grbl32 that is only for the STM32F103 processor and looks to not be actively worked on. for at least 2 years.

By the way, you aren't only person with a lil gap.  But, like rip van winkle, I also woke up.  lol  see the pico discussion section in a bit.

SienciLabs, 2024-03-21:
@phil-barrett thanks for clearing this path ahead of me! I just now found this thread so I'd already retrace the png myself but I went with the same logo on the SuperLongBoard silk too. I'm a fan of it, just because it's simple and effective. Also put it on the box design. I'm not much of an overthinker so if there's still conversation to be had on how Terje or others would like to tailor the official perception of grblHAL in the community then I'm happy to change over if new designs emerge
[image]
[image]

phil-barrett, 2024-03-23:
Not much to say.  Glad to see people using what I made.  Any/everyone is free to share their thoughts on it.

terjeio, 2024-03-25:
FYI [this is a svg](https://github.com/grblHAL/.github/blob/main/media/grblHAL.svg) of the logo.

For those who want to know, the author of the [WebUI](https://github.com/luc-github/ESP3D-webui) selected this logo when he added grblHAL support.

closed 2024-10-30 by mnesarco