grblHAL/core #4: Error Compiling for Teensy4.1

theomajigga, 2021-03-17:
[image]

Can't get passed this error.

terjeio, 2021-03-17:
Did you install [Teensyduino](https://www.pjrc.com/teensy/td_download.html)? It is [required](https://github.com/grblHAL/iMXRT1062).

theomajigga, 2021-03-17:
Yes, same error

terjeio, 2021-03-17:
Sorry, I did not see that you are compiling for the wrong board. You have to select Teensy 4.1 from _Tools > Board: ... > Teensyduino_. If you have already done that then there is something wrong with your installation as the missing file is part of Teensyduino.

theomajigga, 2021-03-17:
Oh christ. What a noob mistake.

theomajigga, 2021-03-17:
What do you know, it worked when I did it right.

closed 2021-03-17 by theomajigga