grblHAL/core #9: Arduino samd serial error

mpiran79, 2021-03-28:
hi, i'm new.
i'm using arduino wht grbl.
now, for a increase speed of engraving, try to use a samd based arduino zero board.
i know previuous issue, i use arduino ide 1.6.11, wht driver samd 1.6.12.
compiling i have an error wht serial.
`undefined reference to `stream_rx_backup(stream_rx_buffer_t*)'`

all error code is:
`Arduino:1.6.11 (Linux), Scheda:"Arduino/Genuino Zero (Native USB Port)"

Opzioni di compilazione cambiate, ricompilo tutto
/home/piro/Arduino/libraries/grblHAL_MKRZERO/src/driver.c: In function 'STEPPER_IRQHandler':
/home/piro/Arduino/libraries/grblHAL_MKRZERO/src/driver.c:1089:5: warning: assignment of read-only location '1107308544u->COUNT32.INTFLAG.bit.MC0' [enabled by default]
     STEPPER_TIMER->COUNT32.INTFLAG.bit.MC0 = 1;
     ^
/home/piro/Arduino/libraries/grblHAL_MKRZERO/src/driver.c: In function 'STEPPULSE_IRQHandler':
/home/piro/Arduino/libraries/grblHAL_MKRZERO/src/driver.c:1096:5: warning: assignment of read-only location '1107307520u->COUNT16.INTFLAG.bit.MC0' [enabled by default]
     STEP_TIMER->COUNT16.INTFLAG.bit.MC0 = 1;
     ^
/home/piro/Arduino/libraries/grblHAL_MKRZERO/src/driver.c: In function 'STEPPULSE_Delayed_IRQHandler':
/home/piro/Arduino/libraries/grblHAL_MKRZERO/src/driver.c:1103:5: warning: assignment of read-only location '1107307520u->COUNT16.INTFLAG.bit.MC0' [enabled by default]
     STEP_TIMER->COUNT16.INTFLAG.bit.MC0 = 1;
     ^
/home/piro/Arduino/libraries/grblHAL_MKRZERO/src/driver.c: In function 'DEBOUNCE_IRQHandler':
/home/piro/Arduino/libraries/grblHAL_MKRZERO/src/driver.c:1118:5: warning: assignment of read-only location '1107305472u->INTFLAG.bit.OVF' [enabled by default]
     DEBOUNCE_TIMER->INTFLAG.bit.OVF = 1;
     ^
/home/piro/Arduino/libraries/grblHAL_MKRZERO/src/serial.c:32:0: warning: "PIN_SERIAL1_RX" redefined [enabled by default]
 #define PIN_SERIAL1_RX (13ul)
 ^
In file included from /home/piro/.arduino15/packages/arduino/hardware/samd/1.6.12/cores/arduino/delay.h:27:0,
                 from /home/piro/.arduino15/packages/arduino/hardware/samd/1.6.12/cores/arduino/Arduino.h:81,
                 from /home/piro/Arduino/libraries/grblHAL_MKRZERO/src/serial.c:26:
/home/piro/.arduino15/packages/arduino/hardware/samd/1.6.12/variants/arduino_zero/variant.h:122:0: note: this is the location of the previous definition
 #define PIN_SERIAL1_RX       (0ul)
 ^
/home/piro/Arduino/libraries/grblHAL_MKRZERO/src/serial.c:33:0: warning: "PIN_SERIAL1_TX" redefined [enabled by default]
 #define PIN_SERIAL1_TX (14ul)
 ^
In file included from /home/piro/.arduino15/packages/arduino/hardware/samd/1.6.12/cores/arduino/delay.h:27:0,
                 from /home/piro/.arduino15/packages/arduino/hardware/samd/1.6.12/cores/arduino/Arduino.h:81,
                 from /home/piro/Arduino/libraries/grblHAL_MKRZERO/src/serial.c:26:
/home/piro/.arduino15/packages/arduino/hardware/samd/1.6.12/variants/arduino_zero/variant.h:123:0: note: this is the location of the previous definition
 #define PIN_SERIAL1_TX       (1ul)
 ^
/home/piro/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/../lib/gcc/arm-none-eabi/4.8.3/../../../../arm-none-eabi/bin/ld: warning: changing start of section .bss by 16 bytes
/home/piro/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/../lib/gcc/arm-none-eabi/4.8.3/../../../../arm-none-eabi/bin/ld: warning: changing start of section .bss by 16 bytes
/home/piro/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/../lib/gcc/arm-none-eabi/4.8.3/../../../../arm-none-eabi/bin/ld: warning: changing start of section .bss by 16 bytes
libraries/grblHAL_MKRZERO/usb_serial.cpp.o: In function `usb_serialSuspendInput':
/home/piro/Arduino/libraries/grblHAL_MKRZERO/src/usb_serial.cpp:180: undefined reference to `stream_rx_suspend(stream_rx_buffer_t*, bool)'
libraries/grblHAL_MKRZERO/usb_serial.cpp.o: In function `usb_execute_realtime':
/home/piro/Arduino/libraries/grblHAL_MKRZERO/src/usb_serial.cpp:206: undefined reference to `stream_rx_backup(stream_rx_buffer_t*)'
collect2: error: ld returned 1 exit status
exit status 1
Errore durante la compilazione per la scheda Arduino/Genuino Zero (Native USB Port).

Questo report potrebbe essere più ricco di informazioni con l'opzione
"Mostra un output dettagliato durante la compilazione"
abilitata in File -> Impostazioni
`

mpiran79, 2021-03-28:
> hi, i'm new.
> i'm using arduino wht grbl.
> now, for a increase speed of engraving, try to use a samd based arduino zero board.
> i know previuous issue, i use arduino ide 1.6.11, wht driver samd 1.6.12. [...]

i use for LASER engraving, for print on wood

terjeio, 2021-03-28:
Add
`#include "grbl/stream.c"`
after
#include "driver.h"
in _usb_serial.cpp_ as it seems cpp code cannot call c code in another compilation unit? I'll commit this as a fix later.

The warning "warning: assignment of read-only location" when writing to bits in the intflag registers is plainly wrong as these are write to clear according to the reference manual. And AFAICT the code works as it should anyway.

mpiran79, 2021-03-29:
ok. fine. i can compile now. 
now i ceck my pinmap and load.
tnx.

closed 2021-03-29 by mpiran79