grblHAL/core #14: MKS SBASE Set current issue

Lopezno68, 2021-04-23:
There is an error in the configuration of the motor current at routine "mks_set_current" in the file mks_sbase.c
The correct line is:
static const uint8_t wiper_registers[] = {**0x00**, 0x01, 0x06, 0x07};

Lopezno68, 2021-04-23:
[image]

terjeio, 2021-04-23:
Oops, likely typo on my part. WIll commit a fix.

It will be great if you can verify that setting the current works as I do not have a MKS board for testing.

Lopezno68, 2021-04-24:
According to the MKS SBASE board schematic, it would also be more accurate if the current setting factor was 100.45f instead of 113.0f.

cmd [1] = (uint8_t) min (255.0f, roundf (current * **100.45f**));

However, the current adjustment functionality works correctly. Nice work with the software, Terjeio

terjeio, 2021-04-30:
The [Marlin driver](https://github.com/makerbase-mks/MKS-SBASE/blob/71348f94b8b8c61ceaaad7788442230d6f71be76/Firmware/Marlin-bugfix-2.0.x/Marlin/src/feature/digipot/digipot_mcp4451.cpp#L34-L61) uses 106.7 as the factor (AFAICT). The schematic I have specifies two different values for Rsense depending on the driver version (DRV8824 vs DRV8825) and R11 = 20K (with an asterix meaning that this might be different on the board?) in the BOM so for me it it is impossible to tell what is correct. So I'll change it to 100.45. I must say I am a bit curious about how you arrived at that factor though.

Lopezno68, 2021-05-01:
Yes, the R11 in my board (DRV8825) is exacty 20K. The Vref impedance in the DRV8825 is superior at 1M1 ohms, then, the Vpot drain current (at the cursor) is negligible. There are 4  50K potenciometers in parallel  and with the R1 they are forming a voltage divisor to obtain 1.269 volts as reference for the 4 potenciometers. Applying electronic mathematics, the value for the potenciometer is  Value=100.45 *Imotor.

[image]

I have measured it on my board with a voltmeter and the theory exactly matches the result.

Lopezno68, 2021-05-01:
it's good to be curious ...

Lopezno68, 2021-05-01:
An other correction. Sorry
[image]

terjeio, 2021-05-02:
> An other correction. Sorry

No problem.

I wonder if the factor should be multiplied with sqrt(2) to set the RMS value as it looks like the current from Equation 1 (in chapter 8.3.2) in the [datasheet](https://www.ti.com/lit/ds/symlink/drv8825.pdf) is the peak current. The Trinamic driver code I've seen does that. However, the [Marlin documentation](https://marlinfw.org/docs/gcode/M907.html) for digipots does not help in making that clear, not even for the [Trinamic setting](https://marlinfw.org/docs/gcode/M906.html) which I assume is the RMS current...

terjeio, 2021-05-05:
Fix committed with your factor.

closed 2022-01-20 by terjeio