grblHAL/core #26: build fails in crossbar for LPC176x

raininja, 2021-06-08:
see the below comment, expansion of PINMODE_CD is failing for some odd reason

https://github.com/grblHAL/core/commit/3a84b58d301f04279268b4ef1045fd6bc0961be5#commitcomment-51905150

terjeio, 2021-06-09:
Workaround comitted, should I add your platformio.ini to the project (or to the core Wiki?) as an example of how to configure a project?

raininja, 2021-06-09:
@terjeio not yet as it doesn't fully build.  there's no open drain pinmode for LPC17xx?

terjeio, 2021-06-09:
> there's no open drain pinmode for LPC17xx?

No, not yet. For now it has been added as an option to some STM32F4xx and STM32F7xx board maps. It takes time to go through all drivers and board maps to see which can benefit from new features, some will only be added on request or if someone provides a PR.

A good thing about the new repo structure is that I can get an idea about which drivers are in use so I can concentrate my efforts around those.

raininja, 2021-06-09:
> No, not yet. For now it has been added as an option to some STM32F4xx and STM32F7xx board maps. It takes time to go through all drivers and board maps to see which can benefit from new features, [...]

Maybe I can attempt to help :)

> A good thing about the new repo structure is that I can get an idea about which drivers are in use so I can concentrate my efforts around those.

The repo structure is much better in it's current form. . . convoluted, yes, but the end result is easier to achieve.  Thanks.

# #

raininja, 2021-06-09:
no need to set OD mode for LPC17xx, it does it automagically 

```
Although the user manual states otherwise, setting the pin to open-drain mode manually in PINMODE_ODx is not required for I2C functions. As soon as you select one of SDA1, SCL1, SDA2, SCL2 for a pin in PINSELx, the corresponding output stage is forced into open-drain mode.
```

https://community.nxp.com/t5/LPC-Microcontrollers/LPC176x-errata/m-p/621501

terjeio, 2021-06-09:
The open drain option was added for external stepper drivers that has optocoupler inputs, typically driven from a 5V supply. Some processors such as STM23 has 5V tolerant outputs when the pin is in OD mode, and it makes sense to add this option for such processors when the board map file is for a board that does not buffer the outputs _and_ allows external drivers.

Determining which processors and boards ths may be relevant for requires a dive into the datasheets/schematics and finding out which processors and boards are relevant to add external drivers for.

raininja, 2021-06-10:
> The open drain option was added for external stepper drivers that has optocoupler inputs, typically driven from a 5V supply. Some processors such as STM23 has 5V tolerant outputs when the pin is in [...]

as I mentioned and documented, LPC176x doesn't require PINMODE_ODx to be set

closed 2022-01-21 by terjeio