grblHAL/core #28: Please stop obliterating git history

brianredbeard, 2021-06-18:
Presently I'm debugging an issue around configuration of the project.

Unfortunately, this is excessively hard to do based on how commit history is being written.  As an example, when evaluating `git blame` on `changelog.md`, there are only two commits present.  Comparing this with https://github.com/terjeio/grblHAL/blame/master/changelog.md reveals the actual change history.

terjeio, 2021-06-19:
I am not sure what you mean by this, is it possible to migrate a file from one repo to another and keep the commit history and I have not done that?

Note that the [https://github.com/terjeio/grblHAL](https://github.com/terjeio/grblHAL) repo is not going to be updated anymore, I will update the readme files to inform about that and link back to this.

DerAndere1, 2021-09-03:
migration while ceeping git history would have been possible with this workflow: https://ao.ms/how-to-split-a-subdirectory-to-a-new-git-repository-and-keep-the-history/ . As this was not done, you now can do this: https://stackoverflow.com/questions/38017693/restore-git-history-from-split-repositories (I did not test it)

terjeio, 2021-09-03:
@DerAndere1 Thanks for the info. However, I am not prepared to spend time doing this.
If a developer wants the access to the whole commit log, and even in more detail, I can give read access (on request) to the master subversion repo.

closed 2021-12-10 by terjeio