grblHAL/core #35: Invert A/B (ganged y) stepper pins?

th3fallen, 2021-07-21:
When using a ganged Y config with an SKR 1.4 it is not possible that i've found to invert the A or B steppers enable pins which makes them inoperative,

Hopefully im missing something but thus far i have not found anything.

th3fallen, 2021-07-22:
or even inverting the pin definitions would work for me but i cant find any examples of how that would be done

th3fallen, 2021-07-22:
```%
; grblHAL
; 1.1f.20210629
; [OPT:VNMSL,35,1024,3,0]
; [NEWOPT:ENUMS,RT+,TC]
; [FIRMWARE:grblHAL]
; [NVS STORAGE:*FLASH]
; [DRIVER:LCP1769]
; [DRIVER VERSION:210703]
; [BOARD:BTT SKR V1.4 Turbo]
;
$N0=
$N1=
; 0 - Step pulse time
$0=10.0
; 1 - Step idle delay
$1=50
; 2 - Step pulse invert
$2=0
; 3 - Step direction invert
$3=0
; 4 - Invert step enable pin(s)
$4=7
; 5 - Invert limit pins
$5=0
; 6 - Invert probe pin
$6=0
; 7 - Disable spindle with zero speed
$7=0
; 10 - Status report options
$10=511
; 11 - Junction deviation
$11=0.010
; 12 - Arc tolerance
$12=0.002
; 13 - Report in inches
$13=0
; 14 - Invert control pins
$14=1
; 15 - Invert coolant pins
$15=0
; 16 - Invert spindle signals
$16=0
; 17 - Pullup disable control pins
$17=0
; 18 - Pullup disable limit pins
$18=0
; 19 - Pullup disable probe pin
$19=0
; 20 - Soft limits enable
$20=0
; 21 - Hard limits enable
$21=1
; 22 - Homing cycle
$22=0
; 23 - Homing direction invert
$23=0
; 24 - Homing locate feed rate
$24=25.0
; 25 - Homing search seek rate
$25=500.0
; 26 - Homing switch debounce delay
$26=250
; 27 - Homing switch pull-off distance
$27=1.000
; 28 - G73 Retract distance
$28=0.100
; 29 - Pulse delay
$29=0.0
; 30 - Maximum spindle speed
$30=1000.000
; 31 - Minimum spindle speed
$31=0.000
; 32 - Mode of operation
$32=0
; 33 - Spindle PWM frequency
$33=5000
; 34 - Spindle PWM off value
$34=0.0
; 35 - Spindle PWM min value
$35=0.0
; 36 - Spindle PWM max value
$36=100.0
; 37 - Steppers deenergize
$37=0
; 39 - Enable legacy RT commands
$39=1
; 40 - Limit jog commands
$40=0
; 43 - Homing passes
$43=1
; 44 - Axes homing, first pass
$44=4
; 45 - Axes homing, second pass
$45=3
; 46 - Axes homing, third pass
$46=0
; 62 - Sleep enable
$62=0
; 63 - Feed hold actions
$63=2
; 64 - Force init alarm
$64=0
; 65 - Probing feed override
$65=0
; 80 - Spindle P-gain
$80=1.000
; 81 - Spindle I-gain
$81=0.010
; 82 - Spindle D-gain
$82=0.000
; 84 - Spindle PID max error
$84=0.000
; 85 - Spindle PID max I error
$85=10.000
; 90 - Spindle sync P-gain
$90=0.000
; 91 - Spindle sync I-gain
$91=0.000
; 92 - Spindle sync D-gain
$92=0.000
; 95 - Spindle sync PID max I error
$95=0.000
; 100 - X-axis travel resolution
$100=250.000
; 101 - Y-axis travel resolution
$101=250.000
; 102 - Z-axis travel resolution
$102=250.000
; 110 - X-axis maximum rate
$110=500.000
; 111 - Y-axis maximum rate
$111=500.000
; 112 - Z-axis maximum rate
$112=500.000
; 120 - X-axis acceleration
$120=1000.000
; 121 - Y-axis acceleration
$121=1000.000
; 122 - Z-axis acceleration
$122=1000.000
; 130 - X-axis maximum travel
$130=750.000
; 131 - Y-axis maximum travel
$131=750.000
; 132 - Z-axis maximum travel
$132=200.000
; 341 - Tool change mode
$341=0
; 342 - Tool change probing distance
$342=30.0
; 343 - Tool change locate feed rate
$343=25.0
; 344 - Tool change search seek rate
$344=200.0
; 345 - Tool change probe pull-off rate
$345=100.0
; 347 - Dual axis length fail
$347=5.0
; 348 - Dual axis length fail min
$348=2.500
; 349 - Dual axis length fail max
$349=25.000
%
``` Entire config for reference

terjeio, 2021-07-24:
I'll check this out when I am "back in business" in a few days time.

terjeio, 2021-07-25:
Ganged axis settings for signal inversions is "inherited" from the main motor, not A or B. E.g. if if Y is ganged the ganged motor takes its settings from Y not B. Currently it is not possible to have different settings for the ganged motor and I am not sure if this is really needed - one would assume the motors and the limit switches have the same specs?

FYI A, B and C are axis letters and only becomes motor designations if number of axes is increased beyond 3. E.g. it is possible to have one ganged axis in a five axis machine if the controller board supports six motors.

th3fallen, 2021-07-25:
Right, the issue was it was not inheriting from the y stepper settings