grblHAL/core #37: modify acceleration according to work

MARIOBASZ, 2021-07-22:
Is it possible to modify the acceleration according to the job (something like the modifiers for feed, rapid, rpm) does it require more aggressive or less aggressive acceleration?
Use trochoidal mechanized, there greater acceleration decreases working time.
I also do pcb plate milling, lower acceleration protects the 0.1mm tip

terjeio, 2021-07-23:
See the [OpenPNP plugin](https://github.com/grblHAL/Plugin_OpenPNP).  M204 is implemented but documentation has not yet been added for it (it is [Marlin style](https://marlinfw.org/docs/gcode/M204.html) with a slightly different meaning of the parameters - grblHAL is not a printer controller).

MARIOBASZ, 2021-07-23:
Excellent!
Momentarily no grblHal (broken usb pin from my arduino by accident, I'll buy another one, I have an esp32 but still don't know how to load the firmware)
I would think not, but I still consult you:
Can you generate an improved "ramp" acceleration profile even if it is not S acceleration management?
    
                                 -------------------- 
                               /                                \
                            /                                     \
                         /                                          \
                 /                                                         \
        /                                                                           \
/                                                                                            \

terjeio, 2021-07-26:
M204 documentation now added.

> Can you generate an improved "ramp" acceleration profile even if it is not S acceleration management?

It might be possible, but not something I want to dig into now - I have a large backlog of private projects. You can give it a try?

MARIOBASZ, 2021-07-27:
Hi, I need help, I am doing something wrong, my Due does not recognize M204 S10
I have moved Plugin_OpenPNP-main to src directory.
Should it be somewhere else? Should I make any changes to grbl.h and grbl.h files?

MARIOBASZ, 2021-07-27:
I have downloaded the new version, however it does not recognize open pnp commands (error 20)

terjeio, 2021-07-27:
I have not added the init call (or the plugin itself as a submodule) to all drivers yet... Sorry about not mentioning that.

Which driver are you using? `openpnp_init();` has to be called at the end of `driver_init()` in _driver.c_.

MARIOBASZ, 2021-07-27:
Thanks for your quick response! I appreciate it. In which controller do you have it implemented to try to copy it?

MARIOBASZ, 2021-07-27:
I haven't gotten it to work.

I have seen that in plugins_init,
if OPENPNP_ENABLE
call openpnp_init

I need to define somewhere Plugin_OpenPNP-main?
I lack knowledge.

terjeio, 2021-07-27:
Please tell me which driver you are compiling and I will get back with the details needed.

MARIOBASZ, 2021-07-27:
Arduino Due

terjeio, 2021-07-27:
Thanks, add this line:

`#define OPENPNP_ENABLE     1 // OpenPNP plugin. To be completed.`

to your [my_machine.h](https://github.com/grblHAL/SAM3X8E/blob/master/grblHAL_Due/src/my_machine.h) at line 39.

You can use the `$I` command to list information about the driver, the OpenPNP plugin will show up there if active.

_driver.c_ already has the needed code in place.

MARIOBASZ, 2021-07-27:
Thanks a lot!
Now it works, with openpnp_init inside driver_init.
Disadvantages of being nice and helping:
become an annoying fly in your io sender repository

closed 2021-07-28 by MARIOBASZ