grblHAL/core #38: HOMING_FORCE_SET_ORIGIN doesn't seem to be taking effect

5ocworkshop, 2021-07-23:
I have a new teensy based grblhal system up and running.  Great stuff.

I am working my way through the config to get me back to parity on my previous Arduino Mega 6x build and one of the things that I had changed was the default MPos being 0,0,0 after homing.

Homing on my machine goes to the front left corner and I have my tool touch-off plate there as well, and would like this to be MPos 0,0,0.  In config.h there is the define:

// After homing, Grbl will set by default the entire machine space into negative space, as is typical
// for professional CNC machines, regardless of where the limit switches are located. Set this
// define to 1 to force Grbl to always set the machine origin at the homed location despite switch orientation.
#define HOMING_FORCE_SET_ORIGIN // Default disabled. Uncomment to enable.

I have uncommented this, built and flashed but my MPos isn't being reset to 0,0,0 after flashing (this define does do that in grbl-mega).

I also tried:

#define HOMING_FORCE_SET_ORIGIN  1

That did not reset MPos either, I'm still getting:

<Idle|MPos:-1109.000,-927.000,-6.000|Bf:35,1023|FS:0,0>

Any suggestions on what to try next?

5ocworkshop, 2021-07-23:
Here is my build info:

(Checkout was a recursive from yesterday afternoon)

[VER:1.1f.20210629:]
[OPT:VNMSL2,35,1024,3,0]
[NEWOPT:ENUMS,RT+,HOME,ES,TC]
[FIRMWARE:grblHAL]
[NVS STORAGE:*FLASH]
[DRIVER:iMXRT1062]
[DRIVER VERSION:210703]
[DRIVER OPTIONS:USB.2]
[BOARD:GRBLHAL2000 - PRINTNC]

And here are my config strings:

$0=10.0
$1=25
$2=0
$3=3
$4=0
$5=7
$6=1
$7=0
$10=1535
$11=0.010
$12=0.002
$13=0
$14=70
$15=0
$16=0
$17=0
$18=0
$19=0
$20=1
$21=1
$22=1
$23=3
$24=350.0
$25=1600.0
$26=250
$27=6.000
$28=0.100
$29=0.0
$30=1000.000
$31=0.000
$32=0
$33=5000.0
$34=0.0
$35=0.0
$36=100.0
$37=0
$39=1
$40=0
$43=1
$44=4
$45=3
$46=0
$62=0
$63=2
$64=0
$65=0
$80=1.000
$81=0.010
$82=0.000
$84=0.000
$85=10.000
$90=0.000
$91=0.000
$92=0.000
$95=0.000
$100=100.000
$101=100.000
$102=250.000
$110=9000.000
$111=9000.000
$112=2032.000
$120=2000.000
$121=1600.000
$122=250.000
$130=1115.000
$131=933.000
$132=138.000
$170=0.000
$171=0.000
$172=0.000
$341=0
$342=30.0
$343=25.0
$344=200.0
$345=100.0
$347=5.0
$348=2.500
$349=25.000

terjeio, 2021-07-23:
`#define HOMING_FORCE_SET_ORIGIN 1` is a default for a setting $22 flag, you have to reset your settings with `$RST=* `after changing it.
Alternatively you can add 8 (for _Set machine origin to 0_) to the current $22 value as grblHAL allows setting this at rutime:
```
$22: Homing cycle as bitfield where setting bit 0 enables the rest:
    0 - Enable (1)
    1 - Enable single axis commands (2)
    2 - Homing on startup required (4)
    3 - Set machine origin to 0 (8)
    4 - Two switches shares one input pin (16)
    5 - Allow manual (32)
    6 - Override locks (64)
    7 - Keep homed status on reset (128)
```

If you are using Windows [ioSender](https://github.com/terjeio/ioSender) has support for setting these flags from the settings UI:

[image]

5ocworkshop, 2021-07-23:
Thank you, really helpful response.  I realized reading your comments that I had in fact had to do the same thing on the Mega, I just forgot about reloading from defaults.

Super helpful to have the UI from IOSender to do it as well.  Will confirm at machine later today and update, but anticipate it is fully resolved.

5ocworkshop, 2021-07-25:
I had a chance tonight to set this from the IOSender UI and it worked as expected.  Thank you!.  Closing.

closed 2021-07-25 by 5ocworkshop