grblHAL/core #39: Limit switches on both ends

karoria, 2021-07-23:
Hello wonderful developers of this project..
I want to ask whether I can configure both side limit switches of any Axis separately? This I am asking because right now I am using grbl_esp32 and enabled 2 limit switches on axis but it takes single input out of both switches. This creates problem when any limit is switched and machine is powered on. It forgets which side to home :)
If separate limits can be input for each axis, kindly guide me How. FYI, I am using Teensy4.1 with customized PCB (yet to be designed)
Thanks in advance.

terjeio, 2021-07-24:
The HAL interface supports up to four limit switches per axis (two for min and two for max), but neither drivers nor the core take full advantage of that yet. A bit more code is required but IMO it should not be too hard to add.

karoria, 2021-07-25:
I am not good at coding. I have seen in one of the pin maps both min and max switches. I want to use only 2 switches per axis. If such code is readily available and you can point me to that code, that willbe great help for me. Thanks for sparing your precious time.
Regards,
Ravi

terjeio, 2021-07-28:
The iMXRT1062 driver already has support for max limit switches for X, Y and Z axes. Since you are designing your own PCB I would guess you are making map file for it. So just add mappings  for `X_LIMIT_PIN_MAX`, `Y_LIMIT_PIN_MAX` and` Z_LIMIT_PIN_MAX`.

karoria, 2021-07-28:
Thank you very much @terjeio .

closed 2021-07-28 by karoria