grblHAL/core #46: state = STATE_IDLE after ALARM is asserted

5ocworkshop, 2021-07-26:
I made a surprising discovery tonight.

I was working on the code for flashing the lights when an alarm is raised and it wasn't working at all, or so it appeared.

If you connect to the machine, home and then intentionally trigger a limit sensor I was getting the correct light sequence for homing, idle and the limit event - but then the IOSender UI and the ? status line both claim Alarm:1 but the STATE flag in the system has returned to STATE_IDLE. 

I discovered this, because on a whim I changed my STATE_ALARM to STATE_IDLE in my light code and was surprised to discover the code works fine, the machine state is the unexpected surprise.

I would think it is not expected behaviour to set the STATE_IDLE flag before the Alarm condition is cleared?

5ocworkshop, 2021-07-26:
I believe it the system should be in `STATE_ALARM` while the Alarm is still asserted and if so desired, a plugin could then use the value of `alarm_code_t` to determine more detail about the alarm condition?

terjeio, 2021-07-26:
> I discovered this, because on a whim I changed my STATE_ALARM to STATE_IDLE in my light code and was surprised to discover the code works fine, the machine state is the unexpected surprise.

Are you sure about this? The current state is used for generating the realtime report and it is not changed to idle when I did a quick test here.

> a plugin could then use the value of alarm_code_t to determine more detail about the alarm condition?

Yes, you can find it in `sys.alarm`.

5ocworkshop, 2021-07-26:
I can reliably reproduce the situation here.

I was quite surprised by it, but looking through the code it seems like the alarm area is a bit of a work in process.

What steps did you use to reproduce it that did not confirm my finding?

To be clear, the report line contains the Alarm:1 entry during the entire situation, it's the additional test that I'm doing that uncovered it.  Perhaps I have a mistake in my code.  I'm on a different PC at the moment but I will post the code for your comment.

terjeio, 2021-07-26:
> it seems like the alarm area is a bit of a work in process.

I have no plans to change the alarm handling, the only thing that annoys me is that I cannot permanently switch on reporting the alarm code as part of the state in the real time report since this might break some senders.

> What steps did you use to reproduce it that did not confirm my finding?

I added a breakpoint at the start of the `state_set()` function in _state_machine.c_, it was only called once for setting the alarm state when I triggered a limit switch. Also, the state variable is encapsulated in _state_machine.c_, the only way to change it is by calling `state_set()` and to get it `state_get()`  has to be called. How the real-time report can then report an alarm when the state is idle I do not understand.

5ocworkshop, 2021-07-26:
Thanks for the background.  I'll be working on this again today.  It sounds like it must be an error in my code.  I need to double check how I am getting the status.

5ocworkshop, 2021-07-27:
Looks like it was an error in my handling of the state.  I believe it is working correctly and I have updated my code.  Sorry for the, er, false alarm. :) Closing.

closed 2021-07-27 by 5ocworkshop