grblHAL/core #47: PlatformIO updated teensy platform from 4.12 to 4.13 and now getting compile errors from the SD modules

5ocworkshop, 2021-07-27:
PlatformIO prompted me to update my teensy platform from 4.12 to 4.13.

Now I am getting errors on my compile that I wasn't getting previously (haven't updated any files and don't use the SD module).

I see there is some history around isssues with the SD module compiling for the teensy but since it pretty much worked out of the box before, I'm a bit confused what has changed or how to resolve it.

The errors:

Compiling .pio\build\teensy41\lib9f7\uSDFS\utility\sd_spi.c.o
In file included from .pio\libdeps\teensy41\MSC-non-blocking\src\msc.h:31:0,
                 from .pio\libdeps\teensy41\MSC-non-blocking\src\MassStorageHost.cpp:31:
C:\Users\xxxxx\.platformio\packages\framework-arduinoteensy\libraries\USBHost_t36/USBHost_t36.h:2045:2: error: 'msSCSICapacity_t' does not name a type
  msSCSICapacity_t msCapacity;
  ^
C:\Users\xxxxx\.platformio\packages\framework-arduinoteensy\libraries\USBHost_t36/USBHost_t36.h:2046:2: error: 'msInquiryResponse_t' does not name a type  
  msInquiryResponse_t msInquiry;
  ^
C:\Users\xxxxx\.platformio\packages\framework-arduinoteensy\libraries\USBHost_t36/USBHost_t36.h:2047:2: error: 'msRequestSenseResponse_t' does not name a type
  msRequestSenseResponse_t msSense;
  ^
C:\Users\xxxxx\.platformio\packages\framework-arduinoteensy\libraries\USBHost_t36/USBHost_t36.h:2048:2: error: 'msDriveInfo_t' does not name a type        
  msDriveInfo_t msDriveInfo;
  ^
C:\Users\xxxxx\.platformio\packages\framework-arduinoteensy\libraries\USBHost_t36/USBHost_t36.h:2067:31: error: 'msSCSICapacity_t' has not been declared   
  uint8_t msReadDeviceCapacity(msSCSICapacity_t * const Capacity);
                               ^
compilation terminated due to -fmax-errors=5.
In file included from .pio\libdeps\teensy41\MSC-non-blocking\src\msc.h:31:0,
                 from .pio\libdeps\teensy41\MSC-non-blocking\src\MassStorageDriver.cpp:29:
C:\Users\xxxxx\.platformio\packages\framework-arduinoteensy\libraries\USBHost_t36/USBHost_t36.h:2045:2: error: 'msSCSICapacity_t' does not name a type     
  msSCSICapacity_t msCapacity;
  ^
C:\Users\xxxxx\.platformio\packages\framework-arduinoteensy\libraries\USBHost_t36/USBHost_t36.h:2046:2: error: 'msInquiryResponse_t' does not name a type  
  msInquiryResponse_t msInquiry;
  ^
C:\Users\xxxxx\.platformio\packages\framework-arduinoteensy\libraries\USBHost_t36/USBHost_t36.h:2047:2: error: 'msRequestSenseResponse_t' does not name a type
  msRequestSenseResponse_t msSense;
  ^
C:\Users\xxxxx\.platformio\packages\framework-arduinoteensy\libraries\USBHost_t36/USBHost_t36.h:2048:2: error: 'msDriveInfo_t' does not name a type        
  msDriveInfo_t msDriveInfo;
  ^
C:\Users\xxxxx\.platformio\packages\framework-arduinoteensy\libraries\USBHost_t36/USBHost_t36.h:2067:31: error: 'msSCSICapacity_t' has not been declared   
  uint8_t msReadDeviceCapacity(msSCSICapacity_t * const Capacity);
                               ^
compilation terminated due to -fmax-errors=5.
Compiling .pio\build\teensy41\libd78\lwip\api\api_lib.c.o
In file included from .pio\libdeps\teensy41\MSC-non-blocking\src/msc.h:31:0,
                 from .pio\libdeps\teensy41\uSDFS\src\utility\sd_msc.cpp:41:
C:\Users\xxxxx\.platformio\packages\framework-arduinoteensy\libraries\USBHost_t36/USBHost_t36.h:2045:2: error: 'msSCSICapacity_t' does not name a type     
  msSCSICapacity_t msCapacity;
  ^
C:\Users\xxxxx\.platformio\packages\framework-arduinoteensy\libraries\USBHost_t36/USBHost_t36.h:2046:2: error: 'msInquiryResponse_t' does not name a type
  msInquiryResponse_t msInquiry;
  ^
C:\Users\xxxxx\.platformio\packages\framework-arduinoteensy\libraries\USBHost_t36/USBHost_t36.h:2047:2: error: 'msRequestSenseResponse_t' does not name a type
  msRequestSenseResponse_t msSense;
  ^
C:\Users\xxxxx\.platformio\packages\framework-arduinoteensy\libraries\USBHost_t36/USBHost_t36.h:2048:2: error: 'msDriveInfo_t' does not name a type        
  msDriveInfo_t msDriveInfo;
  ^
C:\Users\xxxxx\.platformio\packages\framework-arduinoteensy\libraries\USBHost_t36/USBHost_t36.h:2067:31: error: 'msSCSICapacity_t' has not been declared   
  uint8_t msReadDeviceCapacity(msSCSICapacity_t * const Capacity);
                               ^
compilation terminated due to -fmax-errors=5.
*** [.pio\build\teensy41\lib5ae\MSC-non-blocking\MassStorageHost.cpp.o] Error 1
*** [.pio\build\teensy41\lib5ae\MSC-non-blocking\MassStorageDriver.cpp.o] Error 1
*** [.pio\build\teensy41\lib9f7\uSDFS\utility\sd_msc.cpp.o] Error 1
=============================================================== [FAILED] Took 4.48 seconds ===============================================================

5ocworkshop, 2021-07-27:
It appears this new update was just pushed in the last 12 hours and revs Teensyduino to 1.54 which I believe is the likely source of the conflict based on reading the notes here.

I'm investigating how to force PlatformIO back to 4.12 version of their teensy package.

5ocworkshop, 2021-07-27:
PlatformIO IDE > PIO Home > Platforms > Embedded > Teensy > Changelog

Near the top of the page you can choose to install 4.12.

Then you need to go in to:

PIO > Quick Access > Projects & Configurations > Choose your grblHAL project > teensy41 (tab on far right)

In the 'platform' field append @4.12 so the field reads:
${common_teensy.platform}@4.12

Make clean and try building again, you should see the Platform declared at the top of the build window output as 4.12 instead of 4.13.

ArunKalahasti, 2021-07-28:
I received a similar error and was also able to resolve it by reverting to the 4.12 version.

```
In file included from .pio\libdeps\teensy41\MSC-non-blocking\src\msc.h:31:0,
                 from .pio\libdeps\teensy41\MSC-non-blocking\src\MassStorageDriver.cpp:29:
C:\Users\xxxxx\.platformio\packages\framework-arduinoteensy\libraries\USBHost_t36/USBHost_t36.h:2045:2: error: 'msSCSICapacity_t' does not name a type
  msSCSICapacity_t msCapacity;
  ^
C:\Users\xxxxx\.platformio\packages\framework-arduinoteensy\libraries\USBHost_t36/USBHost_t36.h:2046:2: error: 'msInquiryResponse_t' does not name a type
  msInquiryResponse_t msInquiry;
  ^
C:\Users\xxxxx\.platformio\packages\framework-arduinoteensy\libraries\USBHost_t36/USBHost_t36.h:2047:2: error: 'msRequestSenseResponse_t' does not name a type
  msRequestSenseResponse_t msSense;
  ^
C:\Users\xxxxx\.platformio\packages\framework-arduinoteensy\libraries\USBHost_t36/USBHost_t36.h:2048:2: error: 'msDriveInfo_t' does not name a type
  msDriveInfo_t msDriveInfo;
  ^
Compiling .pio\build\teensy41\lib13a\uSDFS\ffsystem.c.o
C:\Users\xxxxx\.platformio\packages\framework-arduinoteensy\libraries\USBHost_t36/USBHost_t36.h:2067:31: error: 'msSCSICapacity_t' has not been declared
Compiling .pio\build\teensy41\lib13a\uSDFS\ffunicode.c.o
  uint8_t msReadDeviceCapacity(msSCSICapacity_t * const Capacity);
                               ^
compilation terminated due to -fmax-errors=5.
Compiling .pio\build\teensy41\lib13a\uSDFS\utility\ff_utils.c.o
*** [.pio\build\teensy41\lib57a\MSC-non-blocking\MassStorageDriver.cpp.o] Error 1
In file included from .pio\libdeps\teensy41\MSC-non-blocking\src\msc.h:31:0,
                 from .pio\libdeps\teensy41\MSC-non-blocking\src\MassStorageHost.cpp:31:
C:\Users\xxxxx\.platformio\packages\framework-arduinoteensy\libraries\USBHost_t36/USBHost_t36.h:2045:2: error: 'msSCSICapacity_t' does not name a type
  msSCSICapacity_t msCapacity;
  ^
C:\Users\xxxxx\.platformio\packages\framework-arduinoteensy\libraries\USBHost_t36/USBHost_t36.h:2046:2: error: 'msInquiryResponse_t' does not name a type
  msInquiryResponse_t msInquiry;
  ^
C:\Users\xxxxx\.platformio\packages\framework-arduinoteensy\libraries\USBHost_t36/USBHost_t36.h:2047:2: error: 'msRequestSenseResponse_t' does not name a type
  msRequestSenseResponse_t msSense;
  ^
C:\Users\xxxxx\.platformio\packages\framework-arduinoteensy\libraries\USBHost_t36/USBHost_t36.h:2048:2: error: 'msDriveInfo_t' does not name a type
  msDriveInfo_t msDriveInfo;
  ^
C:\Users\xxxxx\.platformio\packages\framework-arduinoteensy\libraries\USBHost_t36/USBHost_t36.h:2067:31: error: 'msSCSICapacity_t' has not been declared
  uint8_t msReadDeviceCapacity(msSCSICapacity_t * const Capacity);
                               ^
compilation terminated due to -fmax-errors=5.
*** [.pio\build\teensy41\lib57a\MSC-non-blocking\MassStorageHost.cpp.o] Error 1
========================================================================================================================================= [FAILED] Took 7.88 seconds =========================================================================================================================================
```

karoria, 2021-09-28:
Started from scratch and followed all the instructions regarding libraries and patches (including some #define changes inside files). Now I feel, I am very near to destination. Following are the last lines of build process:

Archiving .pio/build/teensy41/libFrameworkArduino.a
Indexing .pio/build/teensy41/libFrameworkArduino.a
Linking .pio/build/teensy41/firmware.elf
.pio/build/teensy41/libd3a/libuSDFS.a(diskio.c.o): In function `disk_status':
/Users/karoria/Documents/GitHub/iMXRT1062/grblHAL_Teensy4/.pio/libdeps/teensy41/uSDFS/src/diskio.c:36: undefined reference to `MSC_disk_status'
/Users/karoria/Documents/GitHub/iMXRT1062/grblHAL_Teensy4/.pio/libdeps/teensy41/uSDFS/src/diskio.c:33: undefined reference to `MSC_disk_status'
.pio/build/teensy41/libd3a/libuSDFS.a(diskio.c.o): In function `disk_initialize':
/Users/karoria/Documents/GitHub/iMXRT1062/grblHAL_Teensy4/.pio/libdeps/teensy41/uSDFS/src/diskio.c:79: undefined reference to `MSC_disk_initialize'
/Users/karoria/Documents/GitHub/iMXRT1062/grblHAL_Teensy4/.pio/libdeps/teensy41/uSDFS/src/diskio.c:72: undefined reference to `MSC_disk_initialize'
.pio/build/teensy41/libd3a/libuSDFS.a(diskio.c.o): In function `disk_read':
/Users/karoria/Documents/GitHub/iMXRT1062/grblHAL_Teensy4/.pio/libdeps/teensy41/uSDFS/src/diskio.c:133: undefined reference to `MSC_disk_read'
/Users/karoria/Documents/GitHub/iMXRT1062/grblHAL_Teensy4/.pio/libdeps/teensy41/uSDFS/src/diskio.c:124: undefined reference to `MSC_disk_read'
.pio/build/teensy41/libd3a/libuSDFS.a(diskio.c.o): In function `disk_write':
/Users/karoria/Documents/GitHub/iMXRT1062/grblHAL_Teensy4/.pio/libdeps/teensy41/uSDFS/src/diskio.c:192: undefined reference to `MSC_disk_write'
/Users/karoria/Documents/GitHub/iMXRT1062/grblHAL_Teensy4/.pio/libdeps/teensy41/uSDFS/src/diskio.c:183: undefined reference to `MSC_disk_write'
.pio/build/teensy41/libd3a/libuSDFS.a(diskio.c.o): In function `disk_ioctl':
/Users/karoria/Documents/GitHub/iMXRT1062/grblHAL_Teensy4/.pio/libdeps/teensy41/uSDFS/src/diskio.c:353: undefined reference to `MSC_ioctl'
/Users/karoria/Documents/GitHub/iMXRT1062/grblHAL_Teensy4/.pio/libdeps/teensy41/uSDFS/src/diskio.c:346: undefined reference to `MSC_ioctl'
collect2: error: ld returned 1 exit status
*** [.pio/build/teensy41/firmware.elf] Error 1
============================== [FAILED] Took 60.54 seconds ==============================

Environment    Status    Duration
-------------  --------  ------------
teensy41       FAILED    00:01:00.536
========================= 1 failed, 0 succeeded in 00:01:00.536 =========================
The terminal process "platformio 'run'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.

terjeio, 2021-09-28:
Do you have an old version of the uSDFS library? Please check and potentially change a setting in the library as mentioned [here](https://github.com/grblHAL/iMXRT1062#sd-card-plugin).

karoria, 2021-09-29:
OK. I was using uSDFS library which was automatically fetched by PIO. changed that to the suggested in link. Still no luck but coming closer:

> Executing task: platformio run <

Processing teensy41 (board: teensy41; platform: teensy@4.12; framework: arduino)
-------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/teensy/teensy41.html
PLATFORM: Teensy (4.12.0) > Teensy 4.1
HARDWARE: IMXRT1062 600MHz, 512KB RAM, 7.75MB Flash
DEBUG: Current (jlink) External (jlink)
PACKAGES: 
 - framework-arduinoteensy 1.153.0 (1.53) 
 - toolchain-gccarmnoneeabi 1.50401.190816 (5.4.1)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 93 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <uSDFS> 1.1.2+sha.52b0ae3
|   |-- <MSC-non-blocking> 1.0.0+sha.5a2e6f4
|   |   |-- <USBHost_t36> 0.1
|-- <MSC-non-blocking> 1.0.0+sha.5a2e6f4
|   |-- <USBHost_t36> 0.1
|-- <lwip> 2.0.2+sha.58c266f
|-- <lwip> 2.0.2
Building in release mode
Linking .pio/build/teensy41/firmware.elf
.pio/build/teensy41/src/networking/vfs.c.o: In function `vfs_getcwd':
/Users/karoria/Documents/GitHub/iMXRT1062/grblHAL_Teensy4/src/networking/vfs.c:155: undefined reference to `f_getcwd'
collect2: error: ld returned 1 exit status
*** [.pio/build/teensy41/firmware.elf] Error 1
====================== [FAILED] Took 5.99 seconds ======================

Environment    Status    Duration
-------------  --------  ------------
teensy41       FAILED    00:00:05.988
================= 1 failed, 0 succeeded in 00:00:05.988 =================
The terminal process "platformio 'run'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.

terjeio, 2021-09-29:
Ok, this one is on me - `FF_FS_RPATH` has to be set to 2 in [src/ffconf.h](https://github.com/WMXZ-EU/uSDFS/blob/master/src/ffconf.h#L156-L162) in the uSDFS library. I'll have to add a note about that.

PIO obviously links to  an older version of the uSDFS library since you had to set USE_MSC to 0 as it has been changed to 0 in the [latest version](https://github.com/WMXZ-EU/uSDFS/blob/478e1d5e8167c3e353c0bec32726dd53f43edd80/src/utility/sd_config.h#L58-L59). I do not know if that could be problematic.

karoria, 2021-09-29:
> Ok, this one is on me - `FF_FS_RPATH` has to be set to 2 in [src/ffconf.h](https://github.com/WMXZ-EU/uSDFS/blob/master/src/ffconf.h#L156-L162) in the uSDFS library. I'll have to add a note about [...]

Success!! after above change. Thanks for your valuable time and continuous support. Will play with Teensy4.1 & grblHAL now.