grblHAL/core #53: Req Example of debounce

5ocworkshop, 2021-08-05:
I have a quick question on the use of the `hal.port.wait_on_input` function.

I have a switch connected to Aux_In0 that connects the signal pin to ground when pushed.  The function works correctly when the wait mode is '0' / immediate but I have not been able to get any other condition to work.  Ideally I would like to trigger on low or falling but the board becomes inaccessible if I use them. 

Would you be able to confirm the correct syntax to use the function to read a momentary button with debounce on a falling or low trigger condition?

Thanks,

```
 if (hal.port.wait_on_input(1, 0, 0, 0)) {
        if (inspection_light_on == 0) {
        inspection_light_on = 1;
        }
        else {
            inspection_light_on = 0;
        }
    }
```

terjeio, 2021-08-05:
Which driver version? Please post the `$I` response.

When using `wait_on_input()` for anything else than immediate mode then a reasonable timeout has to be provided. If `wait_on_input()` times out it returns `-1`, you will have to test for that.

In plugin code you may consider registering an [interrupt handler](http://svn.io-engineering.com/grblHAL/html/structio__port__t.html#adae7408106ad2522dbad2db4bad11487) instead, support for this has recently been added to a number of drivers. The function pointer is non-zero if registering is supported.

Note that there is no debounce performed on aux inputs.

5ocworkshop, 2021-08-05:
[VER:1.1f.20210629:]
[OPT:VNMZHS2,35,1024,3,0]
[NEWOPT:ENUMS,RT+,HOME,ES,TC]
[FIRMWARE:grblHAL]
[NVS STORAGE:*FRAM]
[DRIVER:iMXRT1062]
[DRIVER VERSION:210703]
[DRIVER OPTIONS:USB.2]
[BOARD:GRBLHAL2000 - PRINTNC]
[AUX IO:4,0,0,0]
[PLUGIN:RGB Indicator Lights v0.4]

Ah, so I need to implement debounce regardless of the method, that is what I needed to know.

terjeio, 2021-08-06:
>  Ideally I would like to trigger on low or falling but the board becomes inaccessible if I use them.

It crashes or hangs for the duration of the wait? `wait_on_input()` with timeout > 0 blocks until the event or the timeout expires.

> Ah, so I need to implement debounce regardless of the method

Yes, if you really need debounce. Currently the best method will be to register an interrupt handler, when called deregister it and delay, then read the input in immediate mode (and take the neccesary action) and reregister the handler. You may also set a flag instead of deregister/reregister, this to ignore further calls to the handler and/or check the input in your real time handler (IIRC you subscribe to the `on_execute_realtime` event).

A delay (or any code that takes a long time to process) should _not_ be executed in an interrupt handler, use [protocol_enqueue_rt_command()](http://svn.io-engineering.com/grblHAL/html/protocol_8c.html#ac649281d41f9ffced34fcab6bec45cbf) to register a function to be executed (once) from the foreground process instead.

5ocworkshop, 2021-08-06:
Great, thank you.  Very helpful.

terjeio, 2021-10-21:
If you register an interrupt handler for the input pin then set the relevant signal when asserted and call [hal.control.interrupt_callback](https://github.com/grblHAL/core/blob/baf5a36698611fcbb1b394ce79f473ff04f77268/hal.h#L328-L331).

If you want status report to show the correct status hook your code into [hal.control.get_state ](https://github.com/grblHAL/core/blob/baf5a36698611fcbb1b394ce79f473ff04f77268/hal.h#L323-L326). Save away the original fn pointer first (e.g to `get_state_org` as in the example below).

In your replacement read the original status and "or" the input with the original.

```
control_signals_t get state (void):
{
    control_signals_t signals = get_state_org();

    signals.e_stop |= (hal.port.wait_on_input(true, <port>, WaitMode_Immediate, 0.0f)  == 1);

    return signals;
}
```
<port> is the port id you want to use.

The [Bluetooth plugin](https://github.com/grblHAL/Plugins_Bluetooth) uses interrupt for an aux input pin and may serve as an example for how to do that.