grblHAL/core #63: How could I go about modifying the code so that I can only drive stepper motors coordinated?

joonasstenvall, 2021-08-25:
Hi, I'm having problems with accelstepper and it's lack of acceleration when using multiple steppers at the same time. So I was thinking of taking the code straight out of marlin/grbl. How should I do this? I'm reading the code and trying to understand it but it has a lot of excess.

Is there any way to use the motion code easily?
I have saved positions like this in my code
[axis1][axis2][axis3][position1]
[axis1][axis2][axis3][position2]
[axis1][axis2][axis3][position3]

From my limited knowledge, I have determined that grbl can look ahead and plan for the next movements. But I have no idea how I would go about doing this.

Thank you for your time!

terjeio, 2021-08-25:
You have to get planner.c and stepper.c working by adding/substituting the needed code from (many) other files.  Call plan_buffer_line() to queue motion. It will be a lot of work...

If you do not mind dragging in everything you can write a plugin for grblHAL that takes over control and calls mc_line() in motion_control.c to enqueue motion.

accelstepper is a Arduino library? If so the grblHAL codebase is not best suited as is optimized for 32 bit processors. If 8 bit support is required then IMO Grbl is a better choice. Or there might be other libraries available that is made for controlling multiple motors?

joonasstenvall, 2021-08-25:
Accelsteppers coordinated stepper movements don't have acceleration. I'm under the impression that grblHAL uses interrupts? Where is that setuped? Maybe I should look into the plugin line but I'm not too sure about that either...

terjeio, 2021-08-25:
>  I'm under the impression that grblHAL uses interrupts? 

Yes.

> Where is that setuped? 

In the processor specific driver.c code.

joonasstenvall, 2021-08-25:
Would you have a starting point on taking over the controlls from grbl with a plugin? Maybe I could have a while(true) in the plugin code or would that be a bad idea?

terjeio, 2021-08-25:
> Would you have a starting point on taking over the controlls from grbl with a plugin?

Not really - depends on what you want to do. There are some [templates](https://github.com/grblHAL/Templates/tree/master/my_plugin) available and a number of plugins that may give you some ideas. 

> Maybe I could have a while(true) in the plugin code or would that be a bad idea?

I think it should work as long as you are calling mc_line() to execute motions. It will block when the planner buffer is full. If you need input from the serial port for generation motion then it will be a tad more complicated.

closed 2021-09-13 by joonasstenvall