grblHAL/core #66: Automatic Tool Change

everskies, 2021-09-01:
Hey! I'm writing a plugin for a DIY ATC.

When looking through the current tool change code, it doesn't seem like I can easily plug into the logic without modifying at least the tool_change.c logic to account for an ATC plugin (And adding an ATC entry in toolchange_mode_t)

Would this be the correct way to approach this? 
The idea is to have GRBL move the Z axis out of the way, send a command either through UART or I2C to a different board that handles rotating of the tool umbrella, swapping the tool and giving control back to GRBL for probing once done.

I'd want this to result in an eventual pull request with example skeleton code in the near future hence I'm asking if my approach is right!

terjeio, 2021-09-01:
> When looking through the current tool change code, it doesn't seem like I can easily plug into the logic without modifying at least the tool_change.c logic to account for an ATC plugin (And adding an ATC entry in toolchange_mode_t)

> Would this be the correct way to approach this?

No, you should write your plugin to replace the current tool change code by claiming the function pointers early and setting `hal.driver_cap.atc` true.

Here are some code I have played with, a while back now, it may give you ideas of how to approach it:

```
/*
  atc.c - An embedded CNC Controller with rs274/ngc (g-code) support

  Driver code for my Mini Mill ATC, 8 tools arranged in a circle
  A motorized socket wrench is mounted in the center, used for opening/closing the spindle nut

  Part of grblHAL

  Copyright (c) 2018-2020 Terje Io

  Grbl is free software: you can redistribute it and/or modify
  it under the terms of the GNU General Public License as published by
  the Free Software Foundation, either version 3 of the License, or
  (at your option) any later version.

  Grbl is distributed in the hope that it will be useful,
  but WITHOUT ANY WARRANTY; without even the implied warranty of
  MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
  GNU General Public License for more details.

  You should have received a copy of the GNU General Public License
  along with Grbl.  If not, see <http://www.gnu.org/licenses/>.
*/

#include <msp.h>
#include <math.h>
#include <string.h>

#include "grbl/hal.h"
#include "grbl/protocol.h"
#include "grbl/motion_control.h"

#define ATC_I2C_ADDRESS (0x4A)

typedef struct {
    float x;
    float y;
} pos_t;

typedef enum {
    CMD_Version = 0,
    CMD_Motor,
    CMD_Latch,
    CMD_SetCurrent,
    CMD_GetState
} atc_command_t;

typedef union {
    uint8_t value;
    struct {
        uint8_t nut_locked   :1,
                nut_unlocked :1,
                spindle_locked :1,
                spindle_unlocked :1;
    };
} atc_state_t;

typedef enum {
    Motor_Off = 0,
    Motor_CW = 1,
    Motor_CCW = 2
[... 331 lines cut ...]
        return Status_Reset;

    if(!mc_line(previous.values, &plan_data))
        return Status_Reset;

    // Restore coolant and spindle state
    coolant_sync(gc_state->modal.coolant);
    spindle_restore(gc_state->modal.spindle, gc_state->spindle.rpm);

    return Status_OK;
}

void atc_init (void)
{
    if(driver_reset == NULL) {
        driver_reset = hal.driver_reset;
        hal.driver_reset = atc_reset;
    }
    hal.tool.select = atc_tool_select;
    hal.tool.change = atc_tool_change;
}

```

`atc_init()` is to be called as part of driver initialization. `hal.driver_cap.atc` should be set to true in `atc_init()`, this will block the [core tool change code](https://github.com/grblHAL/core/blob/384fda6e3a135f6db1050f55a24e93260528111f/tool_change.c#L379-L382) from registering itself.

everskies, 2021-09-01:
Perfect! Exactly what I needed. Thank you

closed 2022-09-12 by terjeio