grblHAL/core #69: Porting to STM32L476.

XRay3D, 2021-09-06:
What functions and hardware resources need to change or use to porter to STM32L476. I have them more than 100 pieces.

terjeio, 2021-09-06:
If this is a low power version of the STM32F4xx series then the [STM32F4xx driver](https://github.com/grblHAL/STM32F4xx) can be used after adding STM32L476 to the to [the clock tree setup](https://github.com/grblHAL/STM32F4xx/blob/160b56d3ddd59fdd9709b1b5c443e1df9e60136b/Src/main.c#L57-L209)?

You have to consult datasheets and also make a test project for comparing code to find out. If very different you may have to write a new driver - but I would guess at least the STM32F4xx driver can be then used as a template. A pin map file is likely needed as well - do you have bare MCUs or are they soldered on a useable board?

XRay3D, 2021-09-07:
> A pin map file is likely needed as well - do you have bare MCUs or are they soldered on a useable board?

I have NUCLEO-L476 and one pallet of MCUs
I will most likely do my CNC PCB.

closed 2022-09-12 by terjeio