grblHAL/core #70: Interop with CNCjs doesn't seem to be updating the DRO

5ocworkshop, 2021-09-07:
Curious if anyone else here is using CNCjs with grblHAL.

I had used it previously with grblMEGA 5x without any issues.

When I went to run a job (after getting back from 3 weeks of vacation) I decided to try CNCjs (as I have some tool length setting macros I really want to use).  I noticed that the DRO doesn't seem to be updating properly.

When I ran my macros they would start off correctly but then start behaving oddly and that's when I noticed the values on the DRO weren't updated, either for tool length or corner finding.

I opened the console within CNCjs and as soon as I issued the ? status command, the DRO populated, however running the macro again resulted in the same issue.

I'm fairly certainly the frequency of the ? commands being issued is a client issue, but I'm unable to think of any reason that they would work with grblMEGA and not with grblHAL, since the board seems to respond just fine when I issue it by hand.  Are there any situations during normal operation that block responses to the ?.

Any suggestions on where to look greatly appreciated.

5ocworkshop, 2021-09-07:
Just a reminder that I'm using grblHAL on a Teensy4.1 based board (via USB in this case).

terjeio, 2021-09-08:
What are the commands sent to the controller when using these macros?

Note that grblHAL has support for M6, if this is sent to the controller then it will normally enter [tool state](https://github.com/grblHAL/core/wiki/Manual,-semi-automatic-and-automatic-tool-change) - if so this could be confusing the sender?
Try after setting `$341=4` - this switches off M6 handling:
```
$341: Tool change mode:
    0 - Normal
    1 - Manual touch off
    2 - Manual touch off @ G59.3
    3 - Automatic touch off @ G59.3
    4 - Ignore M6
```

5ocworkshop, 2021-09-08:
I'll post the CNCjs macro from the machine later today.

However, the problem is present even when idle or jogging, before running the tool change macro.  I was aware of a problem where, until the first move occured, the DRO wouldn't update (in the past, with grblMEGA) but after that it would be fine.  In this instance, I issue the first ? to update the DRO (which unlocks the job buttons in CNCjs) and then I jog around, and the DRO does not update until I issue another ?.

I have seen some older references to it on the CNCjs discussion but it was resolved multiple versions before the one I'm running, and I wasn't having this with the other board.  Someone people described it happening with Marlin firmware due to an extra echo character or something in the start string.  Details here: https://forum.v1engineering.com/t/problems-using-cnc-js/25404/6

In the case of Marlin, this was the change on the CNCjs side to resolve it, apparently:

"Look for MarlinLineParserResultStart. For me it was line 10065. Edit the regex match to the following:

const r = line.match(/^(?:echo:)?start$/);

You should get a whole lot more output when you connect, and things should work."

Could it be something similar here?

I tried compatibility mode 1 and it didn't resolve.  If I try compatibility mode 2 I get an error (I think it is error 18 on the console, will confirm).

dresco, 2021-09-09:
> Curious if anyone else here is using CNCjs with grblHAL.

I am, also Teensy 4.1 board over USB, but haven't seen this issue. I perform a reset/unlock on first connection, else the DRO panel is greyed out, but no real issues once past that. Am currently away for a day or so, but can check build/config specifics later if there is anything that may help..

5ocworkshop, 2021-09-18:
Following dresco's suggestion of reset, unlock, rehome, I am able to use CNCjs.

closed 2021-09-18 by 5ocworkshop