grblHAL/core #72: Define rotary axes for arduino due

MARIOBASZ, 2021-09-20:
Hello Terjeio. I have downloaded your latest version for Arduino Due. I can't remember or find in which file I should define the axes A_axis, B_axis and C_axis (I have defined in grbl N axis 6).
Can you guide me? Thanks, hug.

terjeio, 2021-09-21:
> I have defined in grbl N axis 6

Here I presume:

https://github.com/grblHAL/core/blob/11640adcd2fa66c39eac75028b640826080f735b/config.h#L38

There is no need to change anything else other than choose a board map in [my_machine.h](https://github.com/grblHAL/SAM3X8E/blob/master/grblHAL_Due/src/my_machine.h) that supports the number of required axes. Be aware that none of the suppiled ones supports more than 5 - and a compile time error should pop up if you configure more than the selected map file supports.
So, for 6 axis support you have to add your own map file with the pin mappings required.

MARIOBASZ, 2021-09-21:
your tinyg2_due_map contains the pinmap for six axes.
In bcnc I visualize the six axes running
In your ioSender, it shows me three axes, but if I load again the version loaded before (two months ago) it shows all six axes.

terjeio, 2021-09-21:
> your tinyg2_due_map contains the pinmap for six axes.

Oops, I have missed that - too many board maps to keep track of...

> In your ioSender, it shows me three axes

That is odd, It shows all 6 axes for me. Could it be that you have compiled with COMPATIBILITY_LEVEL > 0?

ioSender configures the number of axes to show from the $I report, I guess I should try add detection from the first real-time report as well. Setting COMPATIBILITY_LEVEL > 0 disables some of the extensions I have added to grblHAL, including reporting the number of axes defined. 

[image]

MARIOBASZ, 2021-09-21:
Now I can see all 6 axes. I do not know the reason (communication?). Trying to record safety door they look at iosender. I will tell you about problems with safery door in Due. Hugs.

closed 2021-09-21 by MARIOBASZ