grblHAL/core #74: Plasma plugin repeats openpnp

MARIOBASZ, 2021-09-22:
For SAM3X8E, remove the plasma plugin to make openpnp work.
Let driver.c ready with
#ifdef OPENPNP_ENABLE
       openpnp_init ();
#end if
Okay, to just uncomment in may_machine?
Hug.

terjeio, 2021-09-23:
You should not modify driver.c to enable any plugin, enable by uncommenting relevant lines in my_machine.h instead.

Enabling both the plasma and OpenPNP plugin at the same time is a bit odd - what do you want to achieve?
Also, the [plasma plugin](https://github.com/grblHAL/Plugin_plasma#dependencies) requires (and claims) a number of input and output ports, not many boards has the pins (or code available) for that. If not available the plugin will fail to add itself:

[image]

The [OpenPNP](https://github.com/grblHAL/Plugin_OpenPNP) plugin is a bit different when it comes to additional outputs as it does not require (or claim) any. However the `M46` gcode requires at least one to be available (`M46` is just a synonym for the standard `M62` and `M63` commands).

Use the `$I` command to get information about the configuration, plugins will show up in the output if enabled as will then number of extra inputs and outputs available (if any).

Eg: `[AUX IO:3,1,0,0]` is reported when 3 digital inputs, 1 digital output and no analog inputs or outputs are available. Note that any I/O claimed by plugins will not show up here.

Here is the `$I` output when both plugins are enabled and the tinyg board map is selected:

[image]

No plasma plugin or aux I/O shown as the required aux I/O pins are not available.

If you want to add aux I/O to the tinyg map then take a look at the end of the [Ramps map](https://github.com/grblHAL/SAM3X8E/blob/master/grblHAL_Due/src/ramps_1.6_map.h) for how it is done. Note that driver.c only has support for adding a finite number of aux I/O pins.

MARIOBASZ, 2021-09-23:
I didn't want to enable plasma.
Just clone SAM3X8E to my Github desktop.
Result, at least for SAM3X8E:
Openpnp folder appears and also plasma folder:
verify: plasma is just the name, inside it has the openpnp files; remove plasma to resolve the conflict.
I also undo my writing in driver.c
Nice to be able to modify acceleration!
(I have not added pin for M46, only interested in driving acceleration)
In a short time I will use it in my cnc.
Thanks.
PS: Later I will need analog / digital inputs and outputs, I will probably continue to bother.

terjeio, 2021-09-23:
> Openpnp folder appears and also plasma folder:

Sorry, I did not get that as everything was ok on my side (I am keeping the master code locally in Subversion). Fixed now.

closed 2021-09-23 by MARIOBASZ