grblHAL/core #75: Q about expected operation of homing and limit switches

5ocworkshop, 2021-10-05:
I have five limit switches on three inputs to my grblHAL setup.

Xmin, Y1min, Y2min & Y2Max (parallel), Zmax & Zmin (parallel).

I use them for homing to Xmin, Y1min, Y2min (with auto squaring), Zmax and that works properly.

However, when the machine is in operation, if I trigger Y2max at the rear of the machine, I get forced in to a homing cycle.  The problem is, I need to manually move off the Y2max sensor first, if I proceed with the homing cycle, since the machine doesn't know the Y2's are in paralell it assumes I am on the front sensor and crashes in to the physical back of the machine.  This is understandable, and mentioned in the config.h.

However, a more desireable outcome and what I am used to from 8 bit grbl, is the ability to reset and unlock and electrically move off the sensor and then manually choose to re-home.  The problem is I do not have that option right now.  I've tried adjusting config options in a fashion that I believe should achieve this, but I have not had success so I wanted to understand what the logic structure is.  I may just be misconfigured, or is this not possible?

As an alternative, I have A & B limit switch inputs on my controller board, so I may move Zmin and Y2max to their own inputs, but I am hoping for a config fix if possible.

Also, given that we know where the machine is in absolute space when it crashes, I'm wondering if it would be feasible to extend the logic to support parallel switches by deriving which switch is likely tripped based on the absolute machine position.  If I'm at the back of my travel and I trigger Y's limit switch, would it be safe to assume it was the rear switch?  Perhaps have a percent of total travel distance that if you're within that % (5%? 10%) of that absolute location then trigger?

Alternatively, could perhaps an initial config cycle could be run after the switches are set up and initial home is defined, where the machine slowly seeks in each direction and locates the switches and stores their location to a config variable for this purpose?

I should note that I do not currently have soft limits configured.  I need to return to that, but I was having issues where the soft limits were being applied to the WCS and not the machine absolute position, leading to strange limits on travel.  I am in the midst of changing spindles so once I am back up and running I will do more thorough QA on the soft limits piece.

My preferred workflow would be:

1) On *initial* power-on, as a configurable option, to in to falt mode and require homing.  (Already present but want to QA it once I resolve other issues).

2) On limit switch trigger, allow unlock but enter a new state where job motion is enabled but program execution is blocked until the system is reset and re-homed.

This would allow the operator to move off limits or reposition out of a crash in a work piece (something I've also had problems doing when a bit got stuck in a metal cut and my only option was to allow the Z to rip upwards inside the homing sequence and it lifted things right off the wasteboard).

Exiting this jog limited state with a reset would then require a homing sequence (if so configured).

Thoughts?

terjeio, 2021-10-05:
> However, a more desireable outcome and what I am used to from 8 bit grbl, is the ability to reset and unlock and electrically move off the sensor and then manually choose to re-home. The problem is I do not have that option right now.

The "Override locks" flag in $22 is not working? $$=22 output:

```
$22: Homing cycle as bitfield where setting bit 0 enables the rest:
    0 - Enable (1)
    1 - Enable single axis commands (2)
    2 - Homing on startup required (4)
    3 - Set machine origin to 0 (8)
    4 - Two switches shares one input pin (16)
    5 - Allow manual (32)
    6 - Override locks (64)
    7 - Keep homed status on reset (128)

Enables homing cycle. Requires limit switches on axes to be automatically homed.

When `Enable single axis commands` is checked, single axis homing can be performed by $H<axis letter> commands.

When `Allow manual` is checked, axes not homed automatically may be homed manually by $H or $H<axis letter> commands.

`Override locks` is for allowing a soft reset to disable `Homing on startup required`.
ok
```

> Thoughts?

On my Mach 3 based machine I have a limits override switch that allows me to reset the machine and jog off any triggered limit switch while I keep it pressed. I like that and I have added support for it in the core, but only a few (one?) driver has support for it for now. Could that be a solution? It is likely a lot easier to add than the other options you have outlined.

https://github.com/grblHAL/core/blob/71dc5ac3331e3d0d195486c0bfd8605ef3e3f1a7/protocol.c#L107-L114

5ocworkshop, 2021-10-06:
> On my Mach 3 based machine I have a limits override switch that allows me to reset the machine and jog off any triggered limit switch while I keep it pressed. I like that and I have added support [...]

Yes, an override button on an aux pin like this could definitely work for my situation.  That's an elegant way to do it.  What would need to happen for this on the Teensy platforms?

I'll double check my config tomorrow at the machine, but my initial impression is that the override locks command is not functioning in my build.

5ocworkshop, 2021-10-10:
> The "Override locks" flag in $22 is not working? $$=22 output:

Thank you for this.  I went back to using IOSender and I could easily see that this was not set.  I had set it in the firmware but I guess my local config is taking precedence and it wasn't set.

I want to switch off CNCjs and be fully on IOSender, I just need to get my probing macros working, now that I have the soft limits resolved.

Will ask my probing question in the probing thread I started and close this one as resolved.

closed 2021-10-10 by 5ocworkshop