grblHAL/core #86: Bitsensor bsmceo4u-pp

boboxx, 2021-11-15:
Has anyone managed to run grblhal on a Bitsensor bsmceo4u-pp (USB MACH3 100Khz Motion Controller) card? It has a STM32F103RCT6 and it's very cheap!

MeKeCNC, 2021-11-15:
if you want to use grblHAL on the board, you must create a CPU map file for this board, I guess it will be worked, look at this Link. https://hackaday.io/project/167000/logs?sort=oldest

terjeio, 2021-11-15:
The RC version has 256K flash so you should be able to if you create a map file and sort out how to program it.
I did a quick search for it and did not find a schematic so pin mappings has to be sorted out by following the PCB tracks?

closed 2022-09-12 by terjeio

Fusseldieb, 2023-08-22:
Also interested!

terjeio, 2023-08-22:
> Also interested!

Firmware can be built by the [Web Builder](http://svn.io-engineering.com:8080/?driver=STM32F1xx&board=Mach3%20USB%20breakout%20%28BSMCE04U%29).

Fusseldieb, 2023-08-22:
> > Also interested!
> 
> Firmware can be built by the [Web Builder](http://svn.io-engineering.com:8080/?driver=STM32F1xx&board=Mach3%20USB%20breakout%20%28BSMCE04U%29).

I've seen that in the meantime, yes, thank you!

However, is there only a bootloader for it (eg Maple, HID, ...), so I could put other stuff on it?

terjeio, 2023-08-22:
> However, is there another bootloader for it (eg Maple, HID, ...), so I could put other stuff on it?

See [these links](https://duckduckgo.com/?t=ffab&q=STM32F103+bootloader&ia=web) for tips.

mintracer, 2023-08-22:
the question is, how to flash it without the 4 pins... :)

Fusseldieb, 2023-08-22:
> the question is, how to flash it without the 4 pins... :)

Which 4 pins? Just use an FTDI adapter. You can flash it via UART.

terjeio, 2023-08-22:
See the wordpress article for info about flashing, also linked to from the Web Builder (via the board _Homepage_ button).

mintracer, 2023-08-22:
> > the question is, how to flash it without the 4 pins... :)
> 
> Which 4 pins? Just use an FTDI adapter. You can flash it via UART.

this one...

[image]

Fusseldieb, 2023-08-23:
> > > the question is, how to flash it without the 4 pins... :)
> > 
> > 
> > Which 4 pins? Just use an FTDI adapter. You can flash it via UART.
> 
> this one...
> 
> [image]

You'll need a soldering iron and connect the wires to these ports (if you want to flash it via the STLink adapter). You can flash it on the UART1 TX/RX pins too, using a normal FTDI (Serial) adapter, they're like $5. 

But in my case I had to cut traces in order to GROUND BOOT1 and put BOOT0 in 3.3V (to enter flashing mode).

It's not thaat easy, but doable. If you have further questions about this particular process, lemme know.

mintracer, 2023-08-23:
there are quite a lot boards without the traces to the st-link pins... only the newer one have... which traces did you cut?
i have some programmer from my creality ender 3, i think they are ftdi...

FrequentlyFascinated, 2023-09-08:
For what it's worth to those who are having trouble flashing this to a BSMCE04U-PP, I had issues trying to flash the firmware onto one of the versions of this board (8mhz crystal version) using STMCubeProgrammer & ST-LinkV2 interface. However, was successful using the STM32 ST-LINK utility (https://www.st.com/en/development-tools/stsw-link004.html) instead. I was getting a memory read error when trying to connect and then flash, but was able flash the firmware with ST-LINK utility in automatic mode. After completed, connected to USB and was able to successfully connect to the  board using UGS, which detected it as a GRBL1.1F(pretty sure, not at my desktop atm to verify the name)

mintracer, 2023-09-08:
i flashed a green one with a geehy chip, it had the 4 pins... flashing went good, iosender sees all the settings and so on... but sometimes i get a warning that the settings could´t be saved, and randomly the board resets all setting to default. kind a strange thing... 
[image]

terjeio, 2023-10-24:
@Fusseldieb Which MCU? I might be it is one of the fake ones?

terjeio, 2023-10-24:
Hmm, is that a [fake/clone](https://hackaday.com/2020/10/22/stm32-clones-the-good-the-bad-and-the-ugly/) or a recycled chip?
BTW does the reset happen often?

terjeio, 2023-12-15:
@Fusseldieb It seems that some fakes, but not all, loose settings ransomly. My board has a genuine MCU and has yet to fail.

Fusseldieb, 2023-12-20:
> @Fusseldieb It seems that some fakes, but not all, loose settings ransomly. My board has a genuine MCU and has yet to fail.

Hmmm, a bummer, a lot of these boards are clones just because they're cheaper to produce. This means that gtblHAL will not be suited for them, unless someone figures out why that happens.