grblHAL/core #87: Why do you report error 31 to "G28 G91 Z0."? I don't think there is any problem with it

hankecnc, 2021-11-15:
It is the G code generated by the post-processing, and I have been using it in the machining center of the FANUC CNC system, and there is no problem

hankecnc, 2021-11-15:
I remember where you mentioned it, but I forgot, and I don't think it has a problem, it passed smoothly on FANUC

terjeio, 2021-11-15:
Is your machine homed? I should be - but it is not the reason for the error. It could be from a previous command since grblHAL repeats the current error message for all subsequent lines, this to avoid unwanted motion.

Attach the program and I can check it.

hankecnc, 2021-11-15:
The error did come from this
[image]
[_model1.txt](https://github.com/grblHAL/core/files/7538766/_model1.txt)

terjeio, 2021-11-15:
It is due to the G80 on line 6, same error when I try with vanilla Grbl Mega, I'll have to check if this is an inherited bug or if it is a NIST RS274 violation.

terjeio, 2021-11-15:
Seems like an inherited bug to me, this code is wrong?

https://github.com/grblHAL/core/blob/f306bfa5987ca2d5030e1396d3c2cba3425dcf1c/gcode.c#L1925-L1928

should be?
```
        // [G80 Errors]: Axis word are programmed while G80 is active.
        // NOTE: Even non-modal commands or TLO that use axis words will throw this strict error.
        if (axis_words.mask && axis_command != AxisCommand_NonModal) // [No axis words allowed]
            FAIL(Status_GcodeAxisWordsExist);
```

hankecnc, 2021-11-15:
> 对我来说似乎是一个继承的错误，这段代码是错误的？
> 
> https://github.com/grblHAL/core/blob/f306bfa5987ca2d5030e1396d3c2cba3425dcf1c/gcode.c#L1925-L1928
> 
> 应该？
> 
> ```
>         // [G80 Errors]: Axis word are programmed while G80 is active.
>         // NOTE: Even non-modal commands or TLO that use axis words will throw this strict error.
>         if (axis_words.mask && axis_command != AxisCommand_NonModal) // [No axis words allowed]
>             FAIL(Status_GcodeAxisWordsExist);
> ```
I can't answer whether you are wrong, I will explain according to my understanding
 Honestly, I don’t know what G80 does before grblhal throws an error.    

I read linuxcnc's introduction about G80     http://linuxcnc.org/docs/2.8/html/gcode/g-code.html#gcode:g80  

I think the only cases where G80 is wrong are the following:
G98 G81 X-23.44 Y-28.627 Z-47.5 R3. F2506. Q2.969
X-52.644 Y-14.218
X-64.364 Y20.75
G80
X0. Y0.
X-40.732 Y24.401
X-17.292 Y37.85

Because G80 cancels the canned cycle, the subsequent X0. Y0.
X-40.732 Y24.401
X-17.292 Y37.85
Inheritance is missing

hankecnc, 2021-11-15:
For testing code that throws errors
G40 G17 G49 G21 G94 G54 G80
G28 G91 Z0
If Z0 is deleted, grblhal can pass smoothly,
However, G28 is using Z0, there is no missing inheritance

terjeio, 2021-11-15:
> Because G80 cancels the canned cycle, the subsequent X0. Y0.

G80 cancels all motion modes, not only canned cycles.

>X-40.732 Y24.401
X-17.292 Y37.85
Inheritance is missing

This is correct, you have to program a motion mode first as there is none active.

> However, G28 is using Z0, there is no missing inheritance

Yes, G28 uses rapid motions (G0) so there is no inheritance (from modal group G1).
That is why I belive there is a bug in the parser, and the fix I have proposed should be good. You can try it?

hankecnc, 2021-11-16:
ok了

closed 2021-11-16 by hankecnc

muhammadelmogy, 2023-10-31:
Hello, I am having exactly the same error here!, the line **_G28 G91 Z0_** was generated for the post processor 
I am using this post processor:
https://github.com/raenji-sk/grblHAL_Fusion360_Post_Processor

This happens when I use _**"safe retracts = G28"**_  but If I use "Clearance Hight" it removes this line from the G-code and it will not throw any errors
I am not really expert with G-code programming but I never had this issue before "I guess this is because I never used this post processor before!" so I am using very recent build of grblHal so is this still a bug?

terjeio, 2023-10-31:
> this still a bug?

Works for me. Post the all the gcode up to and including the line that errors out.