grblHAL/core #88: hi, hi, Terje, do you have plans to make stm32h7 drivers?

hankecnc, 2021-11-18:
hi, Terje Io, do you have a plan to make stm32h7 driver? I have made grbl circuit board of esp32 for sale, aimed at beginner CNC enthusiasts (they use aluminum plate or aluminum profile to make small desktop cnc), I want to make one more This is a circuit board for professional CNC enthusiasts (they use castings). I think STM32H743VIT6 is very good. This chip is very popular and easy to buy. But I can't program. So do you have a plan to make stm32h7? I will wait

hankecnc, 2021-11-18:
This is the official introduction of STM32H743VI    https://www.st.com/zh/microcontrollers-microprocessors/stm32h743vi.html
Can I mail a STM32H743VIT6 development board from China to Norway? It does not matter, you have time to make it,

hankecnc, 2021-11-18:
My plan is to use Raspberry Pi CM4+candle and grblhal to make a CNC numerical control system for sale. I really hate connecting the computer to use cnc. Raspberry Pi cm4 and grblhal work well.

hankecnc, 2021-11-18:
Can I purchase this development board and mail it to you
[image]
?

hankecnc, 2021-11-18:
In China, most cnc enthusiasts use mach3. Their impression of grbl is still on the 8-bit arduino with poor performance. I think grblhal can completely replace the old mach3 with its high performance, stability and open source.

karoria, 2021-11-18:
You should consider Teensy4.1 as it is the fastest controller in the market now, also flagship of grblHAL, so that you can get proper support also. Except you are finding something which is not available in teensy4.1

hankecnc, 2021-11-18:
> 你应该考虑 Teensy4.1，因为它是目前市场上最快的控制器，也是 grblHAL 的旗舰，这样你也可以获得适当的支持。除非您发现了 teensy4.1 中没有的东西

I cannot buy MIMXRT1062DVJ6B or MIMXRT1062DVJ6A, even if it is available, it can be expensive

hankecnc, 2021-11-18:
Let me first try to make the grblhal board of MIMXRT1062DVJ6B

closed 2021-11-18 by hankecnc

terjeio, 2021-11-19:
> Can I purchase this development board and mail it to you

You can if you want to, I could be nice to have another M7 MCU in the driver set, and that board is debuggable.

FYI The Teensy 4.1 is also a M7 - but a pain to work with as it can only be debugged by writing to a secondary UART or by blinking a LED.
There will be no more drivers from me for boards that does not support proper debugging, it is far too expensive in terms of work...

My address:

Terje Io
Ievegen 120
5917 Rossland
NORWAY

terjeio, 2022-03-10:
[Initial driver](https://github.com/dresco/STM32H7xx) made available by @dresco

hankecnc, 2022-03-10:
> [](https://github.com/dresco/STM32H7xx)@dresco提供的[初始驱动程序](https://github.com/dresco/STM32H7xx)

> [Initial driver](https://github.com/dresco/STM32H7xx) made available by @dresco

God, thank you for remembering me, I'm making an offline version of grblhal, he has a 7 inch screen and an embedded linux running grblhal host computer written in QT.

Demitrius, 2022-06-18:
Hello guys!

Did you manage to develop anything with stm32h7?