grblHAL/core #91: Different pwm pin for laser and spindle

MeKeCNC, 2021-11-21:
Is it possible to have 2 different pwm pin for spindle and laser triggered by the $32 variable?

terjeio, 2021-11-21:
Yes - but you have to add driver support or a driver specific plugin for it. With the [Huanyang spindle plugin](https://github.com/grblHAL/Plugins_spindle/blob/fee15a1b86c5527bfc43acdb2f8d0bc6231f9705/huanyang.c#L325-L360) it is already in place.

MeKeCNC, 2021-11-21:
my expectation is , if $32 is activated then laser pwm pin will be triggered (via M3 Sxxx and M5) otherwise spindle pwm pin will be triggered

terjeio, 2021-11-21:
Yes - that is what is the code linked above does. It checks the $32 setting for laser mode and switches the spindle accordingly.

MeKeCNC, 2021-11-21:
I want to use different pwm pins, BESC spindle and pwm laser with fysetc e4 board, does this board support the plugging you mentioned?

terjeio, 2021-11-21:
No, a board specific plugin or driver code has to be written for that.

terjeio, 2022-09-12:
In later versions of the core it is possible for plugin/board support code to [register additional spindles](http://svn.io-engineering.com/grblHAL/html/spindle__control_8h.html#a7ad081a6d642d08e7a340c7ecc4ca92b). These spindles can then be selected with $32. The [spindle plugin](https://github.com/grblHAL/Plugins_spindle/) also add M104 for selecting which spindle to use without changing the $32 setting.

closed 2022-09-12 by terjeio