grblHAL/core #92: Feature request: Preconfigured PlatformIO support

Domush, 2021-11-27:
I'm coming from Marlin, which compiles in VS Code out of the box, and have no experience using C compilers beyond following the occasional how-to. 

Is there anyone who can create and push a pre-configured platformio.ini for this project with support for the most popular boards? 

I've read the Wiki here on how to compile with PlatformIO but it is super vague, and skips a lot of steps and board-specific items that only those who don't need a Wiki to begin with know of. Making a Wiki how-to meant for pros kind of defeats the purpose of a Wiki how-to.

Is there someone who can add platformio support for those of us who don't live and bleed C, but just want to build a CNC?

For reference, I'm trying to build for an SKR 1.3 board, but I'm hoping a robust platformio.ini implementation won't need to be tweaked per-board, like how the Marlin implementation works: you just change the board ID and hit compile.

Thanks.

dresco, 2021-11-27:
Hi, the PlatformIO section in the core wiki is pretty out of date (was carried over from the old repository structure). 

Have you looked at the readme in the STM32F4 repository? That's a much better starting point..

dresco, 2021-11-29:
@terjeio Do you want to delete the "Using PlatformIO for Arduino development" page from the core Wiki? 

My build instructions in there are outdated now, and I think the subject is much better covered by the README.md files in both STM32F4xx and iMXRT1062 driver repositories...

terjeio, 2021-11-30:
> Do you want to delete the "Using PlatformIO for Arduino development" page from the core Wiki?

Perhaps it should be updated or replaced with (links to?) some generic info?

conqueror433, 2021-12-13:
I totally agree with @dresco It will be very helpful if procedure of compiling can be simplified.

@dresco Have you had success compiling Grbl-HAL to SKR 1.3?  I am also looking to do the same for my laser engraver.

dresco, 2021-12-13:
> @dresco Have you had success compiling Grbl-HAL to SKR 1.3?

Am assuming you meant this for @Domush, rather than me? Afraid I don't have any LPC boards, else I'd give it a go!

conqueror433, 2021-12-13:
Ahh yess. Sorry for that.

tcurdt, 2022-07-11:
platformio support would be super great.

terjeio, 2022-10-18:
A webui for bulding binaries is now underway - check this [discussion and preview](https://github.com/grblHAL/core/discussions/203).

terjeio, 2025-12-23:
The WebBuilder can be used to [output a .ini-file for platformio](https://github.com/grblHAL/core/wiki/Compiling-grblHAL#platformio) for most drivers, except the RP2040, that can be used/modified for local builds.

closed 2025-12-23 by terjeio

terjeio, 2025-12-23:
> What's so special about the RP2040 that it is singled out?

Could be that platformio for the RP2040 was not available at the time I set up the Web Builder? Cannot remember now. Anyway compilation runs in a Docker container, and it works, so why change it?

tcurdt, 2025-12-23:
> so why change it?

As said: I was just curious about the reason.

FWIW: This works for my pio projects on the RP2040

```
[env:pico]
platform = raspberrypi
board = pico
framework = arduino
```