grblHAL/core #100: How to define estop pin and how to invert it

karoria, 2021-12-15:
Hello grblHALians!!
After enabling estop, I define estop pin like this "#define ESTOP_PIN (14u)" bit it doesn't seem to work. Kindly guide me on defining it, as I have not seen it in any exmple map file. Further, how to invert it for NO or NC?
Thanks in advance.

terjeio, 2021-12-15:
It is supported by the iMXRT1062 driver, perhaps a few more. It is implemented by [repurposing the reset input](https://github.com/grblHAL/iMXRT1062/blob/97cf1e012251fb6478bdba9f3ed6f8805075f9f9/grblHAL_Teensy4/src/driver.c#L1009-L1034). Search for `ESTOP_ENABLE` to see the rest of the implementation details.

When available inversion is by a bit in the $14 setting, send `$$=14` for info.

karoria, 2021-12-15:
Does that mean, I have to connect ESTOP with reset input on the controller board and I dont need to define any separate Estop pin?

terjeio, 2021-12-15:
You don't have to, it is possible to have both. But if only one of them you have to define the estop pin instead of the reset pin (with the same pin definition).

karoria, 2021-12-15:
Then why I am not getting it working? Is my syntex correct? for pin definition.

terjeio, 2021-12-15:
> Then why I am not getting it working? Is my syntex correct? 

Where can I see what you have done? You have your own pin map file?

karoria, 2021-12-15:
Yes. I have custom pinmap, where I have defined both reset as well as estop pins on separate pins.

terjeio, 2021-12-15:
>  I have defined both reset as well as estop pins on separate pins

That explains it - _driver.c_ has to be modified to support both at the same time. Check the code I linked to above.
The core can handle both, that is why the inversion bits are different.

karoria, 2021-12-15:
Thanks. I will try it.

closed 2022-09-12 by terjeio