grblHAL/core #101: For writing a host computer, depending on whether there is no way to show the user the G code line currently running in grbl?

hankecnc, 2021-12-16:
[image]
For example, like this, show the user the G code line currently executed by grbl, instead of the line that has been sent out

terjeio, 2021-12-16:
I believe might be possible if you add line numbers to your gcode, enable line numbers in the status report and and have a sender that can use those for updating the display. I have never tried that and it may not be exact due to all the buffering going on.

hankecnc, 2021-12-16:
> I believe might be possible if you add line numbers to your gcode, enable line numbers in the status report and and have a sender that can use those for updating the display. I have never tried that and it may not be exact due to all the buffering going on.

Cool!!! Great grbl! Great grblhal! Great Terje Io! Great Norwegian!

[<Idle|Run|Hold|Jog|Alarm|Door|Check|Home|Sleep|Tool>{:<substatus>}|<WPos:|MPos:><axis positions>{|Bf:<block buffers free>,<RX characters free>}{|Ln:<line number>}{|FS:<feed rate>,<programmed rpm>{,<actual rpm>}}{|PN:<signals>}{|WCO:<axis offsets>}{|WCS:G<coordinate system>}{|Ov:<overrides>}{|A:<accessory status>}{|MPG:<0|1>}{|H:<0|1>{,<axis bitmask>}}{|D:<0|1>}{|Sc:<axisletters>}{|TLR:<0|1>}{|FW:<firmware>}]

hankecnc, 2021-12-16:
😊

closed 2021-12-16 by hankecnc

hankecnc, 2021-12-28:
> 我相信，如果您向 gcode 添加行号，在状态报告中启用行号，并且有一个可以使用这些行号来更新显示的发件人，那么这可能是可能的。我从来没有尝试过，由于所有的缓冲，它可能不准确。

I experimented, he thought it was pretty good

https://user-images.githubusercontent.com/69781416/147567775-bf203c63-5178-4f9a-8493-a15e1ed677ab.mp4