grblHAL/core #103: Encountered a strange thing, why input "G1G91Z1.55F200", the actual machine coordinate is 1.552?

hankecnc, 2021-12-30:
[image]
Enter ""G1G91Z1.55F200" again, it will be correct 3.1, which is 1.55*2=3.
[image]
1

hankecnc, 2021-12-30:
I just tested the old version of grbl, he did not have this problem behavior, it is certain that it is because I modified the configuration of conifg.h

hankecnc, 2021-12-30:
?Why? Why even the default grblhal behaves like this

hankecnc, 2021-12-30:
I tried to re-download grblhal from git and re-download and burn, why even the default grblhal behaves like this?

dresco, 2021-12-30:
> I tried to re-download grblhal from git and re-download and burn, why even the default grblhal behaves like this?

Steps per mm at the default setting of 250? You can't hit 1.55mm exactly with that setting (is parameter $102 for the Z axis)..

hankecnc, 2021-12-30:
> > 我试图从 git 重新下载 grblhal 并重新下载和刻录，为什么默认的 grblhal 也会这样？
> 
> 默认设置为 250 时每毫米步数？您无法使用该设置准确地达到 1.55 毫米（Z 轴的参数 $102）。

Well, this is indeed the problem, it is shocking, I have used grbl for a long time, and the old version of grbl is also the same. I only found this problem today.

closed 2021-12-30 by hankecnc

terjeio, 2021-12-30:
>> I tried to re-download grblhal from git and re-download and burn, why even the default grblhal behaves like this?

> Steps per mm at the default setting of 250? You can't hit 1.55mm exactly with that setting (is parameter $102 for the Z axis)..

Position is stored in whole number of steps (since that is what stepper motors move by) and translated back to actual position for the report.

1.55 * 250 = 387.5 steps > 388 steps sent to the motor. 388 / 250 = 1.552. Use >= 1000 steps/mm for more accurate report of position. Not that it may better represent reality since motor, (ball)screw and other tolerances are not likely to to match. E.g a C7 grade ballscrew can have a deviation of max 0.05 mm over 300mm travel.

hankecnc, 2021-12-30:
> Position is stored in whole number of steps (since that is what stepper motors move by) and translated back to actual position for the report.
>
> 1.55 * 250 = 387.5 steps > 388 steps sent to the [...]

Thank you, okay, this is indeed the problem. It is shocking. Today I tried to add a relative coordinate in the host computer production (using the mechanical coordinate obtained this time-the mechanical coordinate obtained last time = the movement increment), and the result is often No, it was finally discovered that it was the problem.