grblHAL/core #107: Homing on RP2040 Pico board

jafgraphics, 2022-01-12:
Hi, I hope you can cast some light onto this.
I'm having an issue homing on a Pico board. The homing sequence alarms out with alarm 8. 
All parts of the homing process seem to work but not all together in a complete homing cycle. It appears something is setting an alarm between the limit striking and been able to pull off the switch.
I think it is some sort of config issue, see driver map below.

Below are the sequence of events.
First sequence is with the limit switch not pressed when homing started

1. The axis will run onto the limit switch at the seek rate.
2. Debounce delay observed
3. Direction of axis is reversed.
4. Pulses the axis for about 20ms and the alarms out with Alarm 8. 

Second sequence is with limit pressed when homing started

1. The axis moves off the limit switch at seek rate the distance set in Pull Off distance
2. Axis runs onto limit switch at feed rate.
3. Debounce delay observed
4. Direction of axis is reversed.
5. Pulses the axis for about 20ms and the alarms out with Alarm 8. 

Many thanks,
Simon

/*
  my_machine_map.h - driver code for RP2040 ARM processors
*/

#if TRINAMIC_ENABLE
#error Trinamic plugin not supported!
#endif

#if N_ABC_MOTORS > 0
#error "Axis configuration is not supported!"
#endif

// Define step pulse output pins.
#define STEP_PINS_BASE 6 // N_AXIS number of consecutive pins are used by PIO

// Define step direction output pins.
#define DIRECTION_PORT          GPIO_OUTPUT
#define X_DIRECTION_PIN         10
#define Y_DIRECTION_PIN         11
#define Z_DIRECTION_PIN         12
#define DIRECTION_OUTMODE       GPIO_MAP

// Define stepper driver enable/disable output pin.
#define STEPPERS_ENABLE_PORT    GPIO_OUTPUT
#define STEPPERS_ENABLE_PIN     14
#define STEPPERS_ENABLE_MASK    STEPPERS_ENABLE_BIT

// Define homing/hard limit switch input pins.  Currently configured so that X and Z limit pins are shared.
#define LIMIT_PORT                  GPIO_IN
#define X_LIMIT_PIN                 2
#define Y_LIMIT_PIN                 3
#define Z_LIMIT_PIN                 4
#define X_LIMIT_BIT                 (1<<X_LIMIT_PIN)
#define Y_LIMIT_BIT                 (1<<Y_LIMIT_PIN)
#define Z_LIMIT_BIT                 (1<<Z_LIMIT_PIN)
#define LIMIT_MASK                  (X_LIMIT_BIT|Y_LIMIT_BIT|Z_LIMIT_BIT)
#define LIMIT_INMODE 				        GPIO_MAP

// Define ganged axis or A axis step pulse and step direction output pins.
//#if N_ABC_MOTORS > 0
//#define M3_AVAILABLE
//#define M3_STEP_PIN             3
//#define M3_DIRECTION_PIN        (Z_DIRECTION_PIN + 1)
//#define M3_LIMIT_PIN            (Z_LIMIT_PIN + 1)
//#endif

// Define spindle enable and spindle direction output pins.
//#define SPINDLE_ENABLE_PORT     GPIO_OUTPUT
//#define SPINDLE_ENABLE_PIN      16
//#define SPINDLE_DIRECTION_PORT  GPIO_OUTPUT
//#define SPINDLE_DIRECTION_PIN   17

// Define spindle PWM output pin.
#define SPINDLE_PWM_PORT        GPIO_OUTPUT
#define SPINDLE_PWM_PIN         22
#define SPINDLE_PWM_BIT         (1<<SPINDLE_PWM_PIN)

// Define flood and mist coolant enable output pins.
//#define COOLANT_FLOOD_PORT      GPIO_OUTPUT
//#define COOLANT_FLOOD_PIN       26
//#define COOLANT_MIST_PORT       GPIO_OUTPUT
//#define COOLANT_MIST_PIN        27
//#define COOLANT_OUTMODE         GPIO_MAP

// Define user-control controls (cycle start, reset, feed hold) input pins.
#define RESET_PIN               18
#define FEED_HOLD_PIN           19
#define CYCLE_START_PIN         20
#if SAFETY_DOOR_ENABLE
#define SAFETY_DOOR_PIN         21
#endif
#define CONTROL_INMODE          GPIO_MAP

// Define probe switch input pin.
#define PROBE_PIN               0

//#if KEYPAD_ENABLE
//#define KEYPAD_STROBE_PIN       1
//#endif

terjeio, 2022-01-12:
Can you check the status shown for the limit pins in the real-time report when the alarm is triggered?
The pin status is in the `Pn:` element, any axis letter there?

_<Idle|MPos:-102.550,-26.450,0.000|Bf:35,1024|FS:0,0|**Pn:P**>_

jafgraphics, 2022-01-12:
The Pn: is always the same as the axis been homed.
So for the Z axis homing the result is Pn:Z, for the X axis is would be Pn:X  etc
I've also checked with a scope and all control lines are behaving as expected. It seems to me like some timer/interrupt is still enabled during the pull off section?

terjeio, 2022-01-12:
No axis letters should show up in the report unless the corresponding limit pin is engaged.

> It seems to me like some timer/interrupt is still enabled during the pull off section?

The limit pin status is polled during homing, and checked after pull-off. If any is still reported as engaged alarm 8 is raised. So no interrupts/timers are involved.

https://github.com/grblHAL/core/blob/f81fcc49f6e515e8723dd42b18dc4859cd809e2d/limits.c#L387-L389

There is a similar check on start of each sequence, if a limit switch is triggered pulloff is tried and alarm 8 is raised if it fails. This is only performed for auto-squared axes.

 https://github.com/grblHAL/core/blob/f81fcc49f6e515e8723dd42b18dc4859cd809e2d/limits.c#L202-L204

terjeio, 2022-01-12:
Do you have hard limits enabled? If so try with disabled - it could be that the limits interrupt is not disabled during homing as it should. FYI I changed the code for enabling/disabling limits interrupt a few days ago, are you using a recent version?

jafgraphics, 2022-01-15:
I finally tracked down the issue, when using the RP2040 driver the hard limits are always enabled, I've altered the limitsEnable function as follows to solve.

    static void limitsEnable (bool on, bool homing)
    {
        uint32_t i = limit_inputs.n_pins;
    
        on = on && !homing;
    
        do {
            i--;
            if(on)
            {
                pinEnableIRQ(&limit_inputs.pins.inputs[i], limit_inputs.pins.inputs[i].irq_mode);//enable
            }
            else
            {
                pinEnableIRQ(&limit_inputs.pins.inputs[i], IRQ_Mode_None);//disable
            }
        } while(i);
    
    #if TRINAMIC_ENABLE
        trinamic_homing(homing);
    #endif
    }