grblHAL/core #114: Third party plug-ins not compiling

5ocworkshop, 2022-02-09:
I did a fresh clone today and wanted to add my status_light plugin back in to my build.

I see the new wording in plugins.h in the grbl directory and I renamed my code status_light.c and put it in the same directory as driver.c as instructed.  I also renamed my init function to be status_light_init().

When I try and compile, I get the following error:

.pio\build\teensy41\src\driver.c.o: In function `driver_init':
C:\Users\jac1d\Documents\PlatformIO\Projects\2022\iMXRT1062\grblHAL_Teensy4/src\grbl/plugins_init.h:101: undefined reference to `status_light_init'
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\teensy41\firmware.elf] Error 1

I think it is probably a simple path or  other issue, but can't see what is wrong.  Any ideas?

5ocworkshop, 2022-02-09:
I forgot to mention, I did add this to my_machine.h:

#define STATUS_LIGHT_ENABLE 1 // Enable RGB status lights plugin

5ocworkshop, 2022-02-09:
Don't mind me, I've been away from the code.  I had added an IFDEF during the last revision, but with the new way the plugin integrates it was checking the incorrect variable.  All better now.

closed 2022-02-09 by 5ocworkshop

karoria, 2022-02-10:
Is your readme file up to date for the compilation part? I want to use your plugin.