grblHAL/core #116: A small problem that is not a problem, send "N1%", it will error: 1

hankecnc, 2022-02-21:


hankecnc, 2022-02-21:
I think grbl should let him through, what do you think?

terjeio, 2022-02-21:
From NIST RS247NCG section 3.1 (my italics):

"The first non-blank line of a file may contain _nothing but a percent sign_, “%”, possibly surrounded by white space, and later
in the file (normally at the end of the file) there may be a similar line."

So no, `N1%` is not legal G-code.

hankecnc, 2022-02-21:
> 

> From NIST RS247NCG section 3.1 (my italics):
> 
> "The first non-blank line of a file may contain _nothing but a percent sign_, “%”, possibly surrounded by white space, and later in the file (normally at the end of the file) there may be a similar line."
> 
> So no, `N1%` is not legal G-code.
OK, thanks

closed 2022-02-21 by hankecnc