grblHAL/core #130: Undocumented New Realtime Commands

SRFirefox, 2022-03-30:
I'm working on a laser cutter overhaul and found grblHAL to be a fantastic option for replacing my 12 year old control board, but there seem to be a few new useful commands that are undocumented. The ones that I'm interested in are CMD_STOP and CMD_OPTIONAL_STOP_TOGGLE. I can presume the intended behavior from the name and a skim of the code, but I'd like to confirm the intended function of these commands before I try to implement them so I can plan for expected behavior.

If these commands allow me gracefully stop a g-code program without losing position they would greatly simplify a fairly typical use-case for lasers for when power/speeds are adjusted as one is learning a new material. In addition, I would be happy to write or update documentation as I'm working on this laser retrofit.

terjeio, 2022-03-30:
`CMD_STOP` is similar to `CMD_RESET` but less harsh. Position is still lost if `CMD_STOP` is issued during motion, send `CMD_FEED_HOLD` followed by `CMD_STOP` when motion is ended to keep position.

`CMD_OPTIONAL_STOP_TOGGLE `controls the behaviour of `M1,` if on `M1` causes a feed hold if off it is ignored.

petervanderwalt, 2026-10-15:
@terjeio both are covered in /grblhal_docs/content/markdown/04-Reference/01-system-commands-reference.md:559
- CMD_STOP → Ctrl-Y / 0x19, documented as Stop.
- CMD_OPTIONAL_STOP_TOGGLE → 0x88, documented as Optional stop toggle. $O is also documented as the text command toggle.
The related M0/M1 program-stop behaviour is covered in G-code & M-code Reference (line 1132) /grblhal_docs/content/markdown/04-Reference/02-complete-gcode-and-mcode-reference.md:1132).