grblHAL/core #132: The latest grblhal version is getting weird? Error 18

hankecnc, 2022-04-13:
I have F4 and H7, they are different from the past, the specific performance is that the power is in the alarm state, and cannot be unlocked through $x, sending $x will get error 18, sending $H, will get error 5, (no code changes

I can't get the unlocking in the sender software. I need to press reset and unlock several times to make the grblhal in the idle state. I don't know what happened to the grblhal during this time.
[image]

[G59.3:0.000,0.000,0.000]
[G28:0.000,0.000,0.000]
[G30:0.000,0.000,0.000]
[G92:0.000,0.000,0.000]
[TLO:0.000,0.000,0.000]
[PRB:0.000,0.000,0.000:0]
[GC:G0 G54 G17 G21 G90 G94 G49 G98 G50 M5 M9 T0 F0 S0.]
[VER:1.1f.20220325:]
[OPT:VNMSL,35,1024,3,0]
[NEWOPT:ENUMS,RT+,TC,SED]
[FIRMWARE:grblHAL]
[NVS STORAGE:*FLASH]
[DRIVER:STM32H743]
[DRIVER VERSION:220325]
[BOARD:grblHAL reference map]
[AUX IO:3,4,0,0]
$0=10.0
$1=25
$2=0
$3=0
$4=7
$5=0
$6=0
$7=0
$10=511
$11=0.010
$12=0.002
$13=0
$14=0
$15=0
$16=0
$17=0
$18=0
$19=0
$20=0
$21=0
$22=0
$23=0
$24=25.0
$25=500.0
$26=250
$27=1.000
$28=0.100
$29=0.0
$30=1000.000
$31=0.000
$32=0
$33=5000.0
$34=0.0
$35=0.0
$36=100.0
$37=0
$39=1
$40=0
$43=1
$44=4
$45=3
$46=0
$62=0
$63=2
$64=0
$65=0
$100=250.000
$101=250.000
$102=250.000
$110=500.000
$111=500.000
$112=500.000
$120=10.000
$121=10.000
$122=10.000
$130=200.000
$131=200.000
$132=200.000
$341=0
$342=30.0
$343=25.0
$344=200.0
$345=100.0
$370=0
$372=0
$384=0
$X
error:18 - Reset asserted
Grbl1.1f ['$' or '$HELP' for help]
[MSG:'$H'|'$X' to unlock]
$X
error:18 - Reset asserted
[GC:G0 G54 G17 G21 G90 G94 G49 G98 G50 M5 M9 T0 F0 S0.]
[G54:0.000,0.000,0.000]
[G55:0.000,0.000,0.000]
[G56:0.000,0.000,0.000]
[G57:0.000,0.000,0.000]
[G58:0.000,0.000,0.000]
[G59:0.000,0.000,0.000]
[G59.1:0.000,0.000,0.000]
[G59.2:0.000,0.000,0.000]
[G59.3:0.000,0.000,0.000]
[G28:0.000,0.000,0.000]
[G30:0.000,0.000,0.000]
[G92:0.000,0.000,0.000]
[TLO:0.000,0.000,0.000]
[PRB:0.000,0.000,0.000:0]
[GC:G0 G54 G17 G21 G90 G94 G49 G98 G50 M5 M9 T0 F0 S0.]
[VER:1.1f.20220325:]
[OPT:VNMSL,35,1024,3,0]
[NEWOPT:ENUMS,RT+,TC,SED]
[FIRMWARE:grblHAL]
[NVS STORAGE:*FLASH]
[DRIVER:STM32H743]
[DRIVER VERSION:220325]
[BOARD:grblHAL reference map]
[AUX IO:3,4,0,0]
$0=10.0
$1=25
$2=0
$3=0
$4=7
$5=0
$6=0
$7=0
$10=511
$11=0.010
$12=0.002
$13=0
$14=0
$15=0
$16=0
$17=0
$18=0
$19=0
$20=0
$21=0
$22=0
$23=0
$24=25.0
$25=500.0
$26=250
$27=1.000
$28=0.100
$29=0.0
$30=1000.000
$31=0.000
$32=0
$33=5000.0
$34=0.0
$35=0.0
$36=100.0
$37=0
$39=1
$40=0
$43=1
$44=4
$45=3
$46=0
$62=0
$63=2
$64=0
$65=0
$100=250.000
$101=250.000
$102=250.000
$110=500.000
$111=500.000
$112=500.000
$120=10.000
$121=10.000
$122=10.000
$130=200.000
$131=200.000
$132=200.000
$341=0
$342=30.0
$343=25.0
$344=200.0
$345=100.0
$370=0
$372=0
$384=0
[GC:G0 G54 G17 G21 G90 G94 G49 G98 G50 M5 M9 T0 F0 S0.]
[G54:0.000,0.000,0.000]
[G55:0.000,0.000,0.000]
[G56:0.000,0.000,0.000]
[G57:0.000,0.000,0.000]
[G58:0.000,0.000,0.000]
[G59:0.000,0.000,0.000]
[G59.1:0.000,0.000,0.000]
[G59.2:0.000,0.000,0.000]
[G59.3:0.000,0.000,0.000]
[G28:0.000,0.000,0.000]
[G30:0.000,0.000,0.000]
[G92:0.000,0.000,0.000]
[TLO:0.000,0.000,0.000]
[PRB:0.000,0.000,0.000:0]
[GC:G0 G54 G17 G21 G90 G94 G49 G98 G50 M5 M9 T0 F0 S0.]
$X
error:18 - Reset asserted
$X
error:18 - Reset asserted
$X
error:18 - Reset asserted
$X
error:18 - Reset asserted
$X
error:18 - Reset asserted

terjeio, 2022-04-13:
Is the reset pin connected to ground via a NC (normally closed) switch or inverted by setting `$14=1`?

dresco, 2022-04-13:
Out of interest, what are you running the H7 code on? 

Has only been bench tested here on a WeAct MiniSTM32H7xx, and a Nucleo H743ZI.

I still need to come up with an initial map for the WeAct board.. I have been working on SDMMC and Ethernet support in different branches, I'll try to get all that updated and pushed to my github repo.

@terjeio - don't know if you saw my post in the F7 repo? I've got a spare Nucleo H7 board I could send to you if that would be useful?

hankecnc, 2022-04-13:
> Out of interest, what are you running the H7 code on?
>
> Has only been bench tested here on a WeAct MiniSTM32H7xx, and a Nucleo H743ZI.
>
> I still need to come up with an initial map for the WeAct [...]

The smallest system board based on h750VBT6, in China, it is very convenient to buy the board,
[image]

hankecnc, 2022-04-13:
> Out of interest, what are you running the H7 code on?
>
> Has only been bench tested here on a WeAct MiniSTM32H7xx, and a Nucleo H743ZI.
>
> I still need to come up with an initial map for the WeAct [...]

I also tried to mail the H743 circuit board to terjeio, but it is too troublesome to send international packages

hankecnc, 2022-04-13:
> $14=1

oh thanks, set $14=1 and it's back to normal

closed 2022-04-13 by hankecnc

hankecnc, 2022-04-13:
> Out of interest, what are you running the H7 code on?
>
> Has only been bench tested here on a WeAct MiniSTM32H7xx, and a Nucleo H743ZI.
>
> I still need to come up with an initial map for the WeAct [...]

hi, I am not a professional programmer, is it difficult to implement G41, G42 tool radius compensation on grblhal? I think grblhal has developed well enough, but the only regret is that he does not support tool radius compensation, so he always only can be called "toy"

terjeio, 2022-04-14:
>  is it difficult to implement G41, G42 tool radius compensation on grblhal?

I believe it is not easy, surely not done in a day or two. A new look-ahead buffer might be needed in addition to the existing planner buffer, and it would require a lot of testing.

I want to see both tool radius compensation and path blending implemented - if I am the one to add it will be later as I want to prioritize my own projects for a while. Perhaps I'll start working on this in the autumn if I am able to travel again.

karoria, 2022-04-14:
Thanks @terjeio,
I am also desperately waiting for tool radius compensation and I know it is going to be difficult till we reach a bug-free code.

andrewmarles, 2022-04-15:
Is cutter compensation perhaps easier to implement in IOSender where we have perfect knowledge of the toolpath?  I know that with GRBLHAL it is much more ideal to do things in the controller, but perhaps this is a useful place to draw a line?

dresco, 2022-04-16:
@hankecnc, I am a bit curious about your board - I think all the H750 only have 128K onboard flash?

fwiw, the H743xI have 2MB flash, and my linker map is using the last 128K sector as EEPROM emulation.

hankecnc, 2022-04-16:
> @hankecnc, I am a bit curious about your board - I think all the H750 only have 128K onboard flash?
> 
> fwiw, the H743xI have 2MB flash, and my linker map is using the last 128K sector as EEPROM emulation.

Except that writing the new grbl system parameters is a bit stuck, grbl replies "ok"" slowly, and other problems have not been found for the time being. It is rumored on the Internet that H750VBT6 is an obsolete product that is essentially unqualified for H743 FLASH.

hankecnc, 2022-04-16:
> @hankecnc, I am a bit curious about your board - I think all the H750 only have 128K onboard flash?
> 
> fwiw, the H743xI have 2MB flash, and my linker map is using the last 128K sector as EEPROM emulation.

st does not have any quality guarantee for the flash behind 128KB? It should be like this

terjeio, 2022-04-16:
> Is cutter compensation perhaps easier to implement in IOSender where we have perfect knowledge of the toolpath?

Perhaps, but I do not want to implement G- and M-codes in the sender. IMO it is better to make them available to all senders by having the code in the controller.

dresco, 2022-04-16:
> > fwiw, the H743xI have 2MB flash, and my linker map is using the last 128K sector as EEPROM emulation.
> 
> st does not have any quality guarantee for the flash behind 128KB? It should be like this

The flash sector size on the H7 is 128KB, so unfortunately that's the smallest erase possible. (I chose to move the settings to the end of flash, as it seemed a bit wasteful to have 128KB for boot, then the 128K for settings, and then the code).

> It is rumored on the Internet that H750VBT6 is an obsolete product that is essentially unqualified for H743 FLASH.

The H750 does seem a bit of an oddball, perhaps that explains it!  If your board is actually saving the settings - then I suppose it must physically have 2MB on there.. 

The WeAct H7xx boards have a couple of 8MB SPI flash chips on them also, so there are more options for program storage. (Their downside is that they tie up a few extra pins for things that we likely won't use, but by my count there are still something like 47 usable pins).