grblHAL/core #136: Insufficient dynamic range on AMASS

andrewmarles, 2022-04-19:
First, I would like to thank @J-Dunn for his work on this issue mentioned here:

https://github.com/Schildkroet/GRBL-Advanced/issues/58

In particular this post:
https://github.com/Schildkroet/GRBL-Advanced/issues/58#issuecomment-871296602

I think it's relevant to open this issue here since the results are interesting for this project.

I think that the point is valid even if he does not wish to engage on this issue.

terjeio, 2022-04-19:
For most of the grblHAL drivers this is not an issue as I use a 32 bit timer whenever possible. In fact I have had to limit the [maximum step time](https://github.com/grblHAL/STM32F4xx/blob/bd2ce508699ccf0a55fd7490976e53b63792897d/Src/driver.c#L479-L483) to avoid what seemed to be occasinally hangs but were overly long step intervals during deceleration.

J-Dunn, 2022-04-19:
>  For most of the grblHAL drivers this is not an issue

Presumably the idea of a HAL is that it works all the time not  whenever possible. 
If you support <32bit architectures in your HAL, the AMASS mod would remove the issue for all h/w archs. 
You comment does point out that it may be better to fix expand the GRBL code to use 32bit counters on STM32 and keep AMASS as it was. 
Is AMASS redundant on 32bit, or does it have other benefits?

terjeio, 2022-04-20:
> Presumably the idea of a HAL is that it works all the time not whenever possible.

grblHAL is designed to be flexible and offers [different features](https://github.com/grblHAL/drivers) for the MCUs/boards supported via the HAL implementation (drivers). Some drivers are limited by the MCU capabilities, some by the effort to needed to implement support for specific features such as those required by plugins.

For MCUs not having a 32 bit timer it is sometimes possible to make use of a [prescaler in the driver code](https://github.com/grblHAL/STM32F1xx/blob/0a6b21565fb8e85d700147c2711ae30368eed6e1/Src/driver.c#L264-L278), I cannot remember now if I have done this for all those drivers and I am lazy to check this now.

> Is AMASS redundant on 32bit, or does it have other benefits?

AFAICT it has [other benefits](https://github.com/grblHAL/core/blob/6d00f743f09db2a07adadf4dc996e3d55de37cdb/stepper.c#L251-L273).

terjeio, 2022-04-20:
> I think it's relevant to open this issue here since the results are interesting for this project.

 It is relevant but IMO slowing the probing speed to get more precise positions is not the necessarily the complete solution. The probe input is polled on every step in the stepper ISR, a problem with this is that if the probe has contact bounce the initial contact might be missed. A possible solution for this is to add interrupt support for the probe pin and use that to trigger a virtual SR latch and then read probe status from the latch output. @[Volksolive](https://github.com/Volksolive) did that for the [RP2040 driver](https://github.com/grblHAL/RP2040/blob/7cd664ebb080cf78592e7ad5bfba59c4cea809ab/driver.c#L982-L1007).
I want to do the same for the other drivers where possible...

J-Dunn, 2022-04-20:
> AFAICT it has [other benefits](https://github.com/grblHAL/core/blob/6d00f743f09db2a07adadf4dc996e3d55de37cdb/stepper.c#L251-L273).

Thanks

> AFAICT it has [other benefits](https://github.com/grblHAL/core/blob/6d00f743f09db2a07adadf4dc996e3d55de37cdb/stepper.c#L251-L273).

so maybe increasing  DR of AMASS is a better solution than probably unnecessarily using 32bit counters which can produce other unexpected consequences.

terjeio, 2026-09-23:
I wonder if it is time to permanently enable [MINIMIZE_PROBE_OVERSHOOT](https://github.com/grblHAL/core/blob/ed9ff69f6b2f0e9b10b8eeaf6920baf88f849e97/stepper.c#L596), ref. [this discussion](https://github.com/grblHAL/core/discussions/133).
It has been available as an option in the Web Builder for some time and no one has reported issues with it.

A while back I also added latching for the probe input when it is wired to an interrupt capable pin, this may reduce the response time if the probe contact is bouncy.