grblHAL/core #140: Feature request : On lathe with spindle sync, get angular position

JoachimF, 2022-05-13:
Hello,

On lathe mode, is it possible to report angular position of the spindle when it's not running, resetable and show it on the IoSender, like C axis for example, for drilling ou slotting.

And on the second part, I think more complicated, to get C axis with stepper, or using the spindle esc as a servo with the sync encoder (it will need A and B signals).

Regards,

Joachim

terjeio, 2022-05-13:
`$SD` reports [spindle encoder data](https://github.com/grblHAL/core/blob/6d00f743f09db2a07adadf4dc996e3d55de37cdb/report.c#L2029-L2049).

To reset call `hal.spindle.reset_data()` or issue a `$SR` command. Note that the function pointer is NULL when not provided by the driver.
The current driver code can not be used to determine angular position if hand turning the spindle, for that a quadrature encoder is needed.

> And on the second part, I think more complicated, to get C axis with stepper, or using the spindle esc as a servo with the sync encoder (it will need A and B signals).

I do not understand what you mean by this.

JoachimF, 2022-05-13:
You're right for the quadrature, I was thinking to turn the spindle by hand only in one way, it's tricky, but with a careful move and a little calculation, determine an angle should be easy, for drilling it's enough. I'll try $SD and $SR.

On CNC lathe, you can use use the spindle chuck as an axis when you have another spindle, (photos : [my lathe](https://photos.app.goo.gl/b5BnpGsJw7LcEgt1A) ) you can use it as a milling machine. [Example](https://www.youtube.com/watch?v=Jab_169Q5os)

If grbl know the position of the spindle, and you can drive the esc (with h-bridge) of the motor, it should be possible to use it as a servo.

Swap one pulse by quadrature encoder is doable ?

Also, on Iosender, if I have more than 3 axis in lathe mode, it crashes.

terjeio, 2022-05-13:
> Swap one pulse by quadrature encoder is doable ?

The encoder API implementation can be changed in the driver so yes. For spindle sync you need 3 inputs, encoder A, B and index pulse.

> Also, on Iosender, if I have more than 3 axis in lathe mode, it crashes.

I'll look into that.

terjeio, 2024-04-16:
> Also, on Iosender, if I have more than 3 axis in lathe mode, it crashes.

This has been fixed.

closed 2024-04-16 by terjeio