grblHAL/core #141: Spindel Speed stimmt nicht

TOKENone7472, 2022-05-15:
Habe das Problem das die Mittlere Spindelgeschwindigkeit nicht stimmt. Ich benutze STM32F446RE und Arduino Shield CNC PWM Signal über Z+. Mittlere Spindelgeschwindigkeit habe ich 1,98V, bei Max Drehzahl 3,29V. 
Die Mittlere Spindelgeschwindigkeit sollte doch ca. 1.65V betragen oder ?
Ich danke schon einmal für Tips und Feedback.

terjeio, 2022-05-15:
[My 446 Nucleo board](https://github.com/terjeio/CNC_Breakout_Nucleo64) outputs spindle voltages that are spot on.
Which board map are you using?
Spindle settings?

[image]

TOKENone7472, 2022-05-15:
[image]
I have 24000 RPM
Spindle PWM frequency 400Hz

terjeio, 2022-05-16:
> Signal über Z+

This is the default pin, D11?

I have verified the output on that pin with your settings, both with scope and multimeter.
Do you measure the voltage when the pin is not connected? Can you check with a scope?

TOKENone7472, 2022-05-17:
problem found. It was my mistake, I built a voltage converter and made a mistake with the capacitor. Everything great now. Thank you for your help.
Greeting TOKEN_one

closed 2022-05-17 by terjeio