grblHAL/core #143: my_machine.h vs CMakeLists.txt

marshr32, 2022-05-25:
I see there is an option to select which board to compile for in both these files.
Do they have to both be set manually? 
Or does my_machine.h set the same board to ON in CMakeLists?

It's just that there is no mention of CmakeLists in the compiling instructions, only my_machine.h 

Many thanks :)

terjeio, 2022-05-26:
_my_machine.h_ contains this note
```
/*
  my_machine.h - configuration for ESP32 processos

  NOTE: only in use if not compiling with cmake (idf.py)

  ...
```

and the [readme this](https://github.com/grblHAL/ESP32#changelognotes).

so it is normally not in use. I guess I should hint at that in the generic compiling instructions.

terjeio, 2024-04-16:
In later versions my_machine.h is now used instead of CMakeLists.txt.

closed 2024-04-16 by terjeio