grblHAL/core #146: XYUV Axes

rcKeith, 2022-06-05:
Hi
Is it possible to set the 3rd and 4th axis to U and V for foam cutting?  I can see in the setting.h there are some options for Group_UAxis and  Group_VAxis but I'm not sure how to configure these.  I have the RP2040 configured and loaded using XYZA but it would be much better for g-code generation software to use XYUV as these seem to be the defacto standard for foam cutting type machines.

Many thanks
Keith

terjeio, 2022-06-16:
> Is it possible to set the 3rd and 4th axis to U and V for foam cutting? 

Not without code changes. There is a fair chance it would work if you change A B C characters to U V W here:

https://github.com/grblHAL/core/blob/45b0e105892d08e76d5140871cfb032d445d2f20/gcode.c#L1113-L1135

and here:

https://github.com/grblHAL/core/blob/45b0e105892d08e76d5140871cfb032d445d2f20/nuts_bolts.c#L60-L68

Can you try this?

> I can see in the setting.h there are some options for Group_UAxis and Group_VAxis but I'm not sure how to configure these.

None of the current drives supports more than 6 axes so you have to remap ABC axes instead.

rcKeith, 2022-06-16:
Hi 
Many thanks, I'll give it a try and let you know
Keith

rcKeith, 2022-06-17:
Hi 
It did work making the changes as you suggested but the only issue is the Pin information coming back still refers to XYZAB  which will cause limit switch problems.  I did look through the code but couldn't see where I could change this.
Thanks
Keith

terjeio, 2022-06-17:
axis_signals_tostring() in report.c

terjeio, 2022-09-27:
The latest build has support for [remapping ABC to UVW](https://github.com/grblHAL/core/blob/357e4ca0778a926f23bb7fbe5c8ba85d41286fcd/config.h#L56-L58).

closed 2022-09-27 by terjeio