grblHAL/core #148: Sync issue between laser output and motor posistion

UniqueLemming, 2022-06-14:
Is there a way to adjust the timing of the spindle pwm output? I would love to delay it relative to the motor signals.

I don't actually think this is an issue with the grblHAL code, but I'm pulling my hair out trying to fix an issue. My laser is firing slightly ahead of when the motors manage to get the laser head to the right spot on my co2 laser. So for cuts I get an overburn at the start posistion and a failure to fully complete the cut at the end. This also happens at sharp corners. It's quite noticable when engraving as well. If I try to fill a shape or engrave an image, each line is offset depending on which direction the head is moving. The offset increases with speed increases. I've made sure to use a large enough overscan in my testing and the head is fully accelerated to the correct speed while this is happening. I've also tried a few different stepper driver options. 
I'm using a BTT skr pro v1.2 board and I've tested it with tmc2209s in UART, tmc2130s in SPI and even old school drv8825s. I get the same result with all of them. The problem exist to the same degree in both the X and Y directions.  The machine moves like butter with the 2130s and since this issue happens at speed, I don't think it's a backlash or other physical problem. I'm really not sure what else I can try.

I'm hoping there might be a way to apply a timing offset to the pwm output for the laser power. It seems a bit unlikely, but I thought I would ask.

My other lasercutter runs grblHAL on a smoothieboard. No sync issues with that one at all and I've been quite happy with it.

terjeio, 2022-06-14:
This is very odd. Have you tried to swap controllers in order to rule out hardware issues?
Implementing a delay will not be easy - especially if you use grayscale engraving, but I cannot understand why that should be needed...

The [PPI plugin](https://github.com/grblHAL/Plugins_laser/) may be of interest to you. Using PPI mode will eliminate overburn in corners - I use it all the time with my laser.

UniqueLemming, 2022-06-14:
I actually originally had one of these https://github.com/bdring/6-Pack_CNC_Controller running FluidNC or grblESP32 and the machine was doing the same thing. I would just believe that I just didn't know how to configure grbl if I didn't have my Smoothieboard laser running so good. I've tried to make things as simple as possible on my machine and even wired the co2 power supply so that it doesn't need the enable signal just to make sure that wasn't messing with things. The pwm signal is all it needs to fire. I guess I'll need to pull out my smoothieboard and try it in there, but that might not be something I can do right away. I actually tried to get another smoothieboard instead of this SKR pro v1.2, but they aren't really available anymore.

I did start looking into the PPI plugin and it does look really interesting. I would love to give it a try, but I wasn't sure how to go about enabling it.

terjeio, 2022-06-15:
> I actually originally had one of these https://github.com/bdring/6-Pack_CNC_Controller running FluidNC or grblESP32 and the machine was doing the same thing.

IMO it is likely that it is a hardware issue then?

> I did start looking into the PPI plugin and it does look really interesting. I would love to give it a try, but I wasn't sure how to go about enabling it.

Only a few drivers has the support code needed. For those who have PPI is enabled in [my_machine.h](https://github.com/grblHAL/STM32F4xx/blob/c5b8e6a0dd225c07200681ab71bc03c10000e16d/Inc/my_machine.h#L67).

UniqueLemming, 2022-06-16:
Okay I did decide to rip my smoothieboard out and try it in that machine. Same problem, so it's obviously not a controller issue. I'll try rewiring my co2 power supply to trigger differently and see what happens. It's currently wired the same as my other lower powered laser, but who knows. Running out of options. I'll stop wasting your time with this though. Thanks a lot for the responses.

Once I get the SKR pro v1.2 back in there I'll see what happens with PPI mode. The STM32F4xx driver does have the option to be enabled in my_machine.h I now see.

closed 2022-06-16 by UniqueLemming