grblHAL/core #149: Is he the founder of grbl? Why are he and terjeio Norwegian? Strange question

hankecnc, 2022-06-14:
Is he the founder of grbl? Why are he and terjeio Norwegian? Are he and terjeio good friends in real life? Strange question

Really great Norwegian!!

[image]

hankecnc, 2022-06-14:
I've looked at the map of Norway and it looks like it's small? It looks like a small town

terjeio, 2022-06-14:
> Is he the founder of grbl?

Yes.

> Why are he and terjeio Norwegian?

By chance?

> Are he and terjeio good friends in real life?

No, I have never met him or corresponded with him. AFAIK he lives near/in the capital Oslo and I live near [Bergen](https://en.wikipedia.org/wiki/Bergen) on the west coast - 450 km apart by road.

> I've looked at the map of Norway and it looks like it's small?

It is a small country but rather long - the (straight) [coastline](https://www.lifeinnorway.net/norway-coast/) is 2650 km...