grblHAL/core #151: Set a pin to high in GRBL

Henrikastro, 2022-06-17:
So, I'm trying to add a button to the sender that I'm using so it triggers a relay in my circuit board and changes the level of a pin to high. I've been doing this by sending something like a "/F" to the Grbl and doing a DigitalWrite in the protocol.c whenever I received that string. Now I'm transitioning to GrblHAL and I've been trying to do it the same way, but with the hal, I have no direct contact with the pins to do that. I've tried to do it using the hal struct but I dont think theres anything that helps me there. If anyone can help me get to the pins or shed a light on a better way of doing this I would really apreciate it.

terjeio, 2022-06-17:
If the driver/board has support for aux outputs you can use [M62-M65](https://www.linuxcnc.org/docs/html/gcode/m-code.html#mcode:m62-m65) - no need to call `hal.port.digital_out()` which by the way [may be a null pointer](http://svn.io-engineering.com/grblHAL/html/structio__port__t.html) if no aux outputs are available (and cause a crash). Check the `$pins` command output, it will have entries like this if so:

```
[PIN:PB15,Aux out 0,P0]
[PIN:PB2,Aux out 1,P1]
[PIN:PA6,Aux out 2,P2]
[PIN:PA5,Aux out 3,P3]
...
```
Note that the id (P or port number) used for interacting with the pin is _not_ the same as the MCU pin number.

If no aux outputs are available you may add them in the board map if you have free pins.

Henrikastro, 2022-06-17:
I'm using an ESP32, I believe it should support M62-M65 but I get an error 20 when trying to run it, should I be enabling it somewhere before using it? Anyways, I think the aux outputs are my way of getting this to work in this case, thank you!

Henrikastro, 2022-06-22:
After adding the `#define HAS_IOPORTS` and `#define AUXOUTPUT0_PIN 21` to my map I was able to use `hal.port.digital_out(0, true)` and then I could control the pin status. Thank you so much!

closed 2022-06-22 by Henrikastro