grblHAL/core #152: Progress with STM32H7

Demitrius, 2022-06-18:
Hello Terje Io and All.

Same question like in #88 

I want to use BTT motherboard with STM32H743VIT6 cpu.
How about progress with STM32H7 ?
May I help you some how?

terjeio, 2022-06-18:
> May I help you some how?

You can by adding a board map for the [STM32H7](https://github.com/dresco/STM32H7xx) driver written by @dresco.

Demitrius, 2022-06-19:
Good. I will try when board arrival. About 1 month.

Also, found this board added to Marlin firmware.
https://github.com/MarlinFirmware/Marlin/pull/24271/commits/e63d69208826bf7fffb86aacd22eba3cfa80c6e5#

dresco, 2022-06-19:
Hi @Demitrius

Is that the BTT SKR 3 board you have on order? I had been wondering about getting one for testing.

Thanks for the link to the Marlin code, will probably be something useful in there..

dresco, 2022-06-21:
For info, have ordered a BTT SKR 3 EZ board with 2209 drivers. Will pull some pin definitions out of the Marlin code and get it up & running when it arrives.

One thing I've noticed from the Marlin code already, for some reason the I2C EEPROM has been put on the SWD pins, so looks like no proper debug on this board :( 

Am hoping to make my own board for the H743, but will realistically be a (northern hemisphere) winter project, so a while away..

Demitrius, 2022-06-21:
Hi @dresco 

Ordered a BTT SKR 3 EZ board with EZ5160 drivers.
Its not good about debugging.

dresco, 2022-07-02:
My board arrived today (that was quick). 
I'll try to find some time this week to create a board map and bring it up for initial testing..

Demitrius, 2022-07-02:
Great! My still on the road.

Demitrius, 2022-07-21:
Hello. Board arrived. Can I do something for you?

dresco, 2022-07-21:
Cool. I've been updating issue #158 - I have my board up and running with the 2209 drivers using software UART.

I plan to pick up a 5160 breakout for testing when they come back into stock. In terms of porting the code to the H7, I think the F4 repo is the best place to start  - as it has Trinamic SPI code for SKR 1.1, SKR 2, and Nucleo boards. 

@terjeio I've just noticed that the SKR 2 board driver has a s/w SPI implementation already - I'd missed that previously. What do you think the best approach is, to try and refactor both h/w and s/w SPI implementations out of the board drivers into a tmc_spi.c or similar?

terjeio, 2022-07-21:
@dresco - refactoring into a _tmc_spi.c_ sounds like a good idea. I can backport to the F4 if you do it for the H7.

dresco, 2022-07-24:
@terjeio  Am making progress with this - but something in the trinamic driver is puzzling me...

Within `on_driver_setup()`, `driver_setup()` is called before `trinamic_drivers_setup()`. This means the GPIO init logic runs before the CS pins have been enumerated, and therefore they never get set as outputs?

https://github.com/grblHAL/Plugins_motor/blob/f371ebcd5fd8e2edf6aa7e77146189eb1c425bde/trinamic.c#L1622-L1632

I can't see how this would work for the existing drivers, or am I missing something? Thanks!

terjeio, 2022-07-24:
This is [where and how all outputs](https://github.com/grblHAL/STM32F4xx/blob/1b37be5beb34a3c8d90b1579f6fccbcf5fc23263/Src/driver.c#L1780-L1807) pins are configured in `driver_setup()` for the F4. Its this what you are looking for?

dresco, 2022-07-24:
From what I see here, the issue is that function is called **before** `hal.enumerate_pins(true, add_cs_pin)` has been executed, so the CS outputs aren't there to be configured?

In the link I posted above, the HAL driver setup is called first (configuring the pins), and then the trinamic driver setup is called (adding the chip select pins as outputs)..

terjeio, 2022-07-24:
> so the CS outputs aren't there to be configured?

I am not sure what you mean by this - they are configured as outputs and set high in `driver_setup()` and then bound to the axes during enumeration. Do you want to add more to the configuration?

dresco, 2022-07-24:
> I am not sure what you mean by this

Ahh - I've just realised my mistake. I was assuming the enumeration had to happen **before** driver_setup() was run in order to configure them, but I now see that is not the case at all.

I had tripped myself up by (a) initially forgetting to add the PinGroup_Motor handling to driver_setup(), and then (b) forgetting to change DIGITAL_OUT to suit the F7/H7 implementation (shifting by the pin number instead of using it directly). 

Had convinced myself it wasn't working for all the wrong reasons, apologies! ;)

dresco, 2022-07-24:
I've pushed my work so far to a new [tmc_spi](https://github.com/dresco/STM32H7xx/tree/tmc_spi) branch on my repo. I don't have the hardware to test yet, but both software and hardware SPI implementations appear to be starting okay.

One thing of note, I've not been able to change the prescaler after the SPI peripheral has been initialised. The writes to the config register don't have any effect (possibly locked if in use?), will look into that...

dresco, 2022-07-24:
> The writes to the config register don't have any effect (possibly locked if in use?), will look into that...

Fixed in latest commit I think, the SPI peripheral just needed to be disabled before writing to the register.

terjeio, 2022-07-24:
If you have time you may want to try [WebUI v3](https://github.com/grblHAL/Plugin_WebUI), it stresses the networking code (http and websocket daemons) as well as the SD card so great for testing. I have added a [new API call to enet.c](https://github.com/grblHAL/STM32F7xx/blob/a4ddcb99fc9bdeb95e921e1b20d039bbb0d5b62b/Src/enet.c#L79-L110) that is required, you'll have to add that as well as update to the latest core.

dresco, 2022-07-24:
> If you have time you may want to try [WebUI v3](https://github.com/grblHAL/Plugin_WebUI)

Cool, will do..

dresco, 2022-08-05:
@Demitrius I've just pushed the TMC5160 support to my [master](https://github.com/dresco/STM32H7xx) branch if you want to test? 

I don't have the 5160 modules to test on my SKR3, but have successfully verified communication with a [TMC5160-BOB](https://www.trinamic.com/support/eval-kits/details/tmc5160-bob/) on my WeAct dev board.

Demitrius, 2022-08-05:
@dresco Good news!

Could you provide me list of task I can do?
Similar:

1 - Connect TMS5160 (EZ my case) to board
2 - Connect USB Board - PC
3 - Need I connect 24V power supply ? (It still on the road)
4 - Install PlatformIO (or other) env to build firmware
5 - Upload firmware to board
6 - See logs (how?)

dresco, 2022-08-07:
@Demitrius 

From my testing with the 5160 breakout, it does need motor power to be supplied before it will respond, so you will need the 24v power supply.

Is probably easiest to build and program with PlatformIO (STMCubeIDE needs another tool for DFU uploads), or I can provide a binary for testing if easier. 

Please note however, there is currently something odd with the PlatformIO builds - for instance SD Card access issues that I don't see on my STMCubeIDE builds. (I haven't spent much effort investigating this yet, as their H7 library is out of date anyway - so am unable to make any PlatformIO Ethernet builds, and was waiting for that to be updated first)..

terjeio, 2022-08-11:
> STMCubeIDE needs another tool for DFU uploads

If you enable binary output that should be usable?
[image]

dresco, 2022-08-11:
> If you enable binary output that should be usable?

Yep, I've added an entry to my "External Tools" menu to call a command line DFU uploader, that seems to work well enough. (Incidentally, the ST32CubeProgrammer CLI also accepts .elf files for DFU upload, but I expect most people would use something like dfu-util instead)..

I think the issues I was seeing with PlatformIO builds are resolved also. I may eventually try to set up a GitHub action to build the PlatformIO firmwares automatically.

dresco, 2022-08-15:
> I may eventually try to set up a GitHub action to build the PlatformIO firmwares automatically.

This turned out to be easier than I expected - is now building automatically on each push to master, and the resulting firmware files are archived in the artifacts for each run.

Am currently building the following, in standard 3-axis configurations;
- BTT SKR 3 with SDCard, USB serial, and TMC 2209 driver support.
- BTT SKR 3 with SDCard, USB serial, and TMC 5160 driver support.
- WeAct MiniSTM32H743 with SDCard and USB serial.
- Nucleo H743ZI with Ethernet and SDCard support (SDCard is required for WebUI & needs a breakout board)

andrewmarles, 2022-08-15:
@dresco did you find any specific documentation to set up the automatic builds?  This is something I would like to do on the F4xx driver.

dresco, 2022-08-15:
> @dresco did you find any specific documentation to set up the automatic builds? This is something I would like to do on the F4xx driver.

@andrewmarles Only from the platformio [docs](https://docs.platformio.org/en/latest/integration/ci/github-actions.html), and looking at other examples online. My config is [here](https://github.com/dresco/STM32H7xx/tree/master/.github/workflows), and I think it should work as-is in the F4 repo. The workflow file(s) just need to go into a .github/workflows folder in the main/master branch.

Am archiving any generated firmware files into the artifact for each run - however these are only stored for 90days from the build. 

Possibly a good next step could be to create a more permanent release action. Perhaps creating tags & releases based off the `hal.driver_version` string? But that's getting more ambitious, and not something I've tried before either..

Demitrius, 2022-09-11:
@dresco early we made successfull test TMC2209 and 5160 drivers with STM32H7. Will you merge changes to master branch?

closed 2023-04-29 by terjeio