grblHAL/core #155: G5 cubic spline

kimstik, 2022-06-28:
https://github.com/grblHAL/core/wiki/Changes-from-grbl-1.1

I think fact that G5 supported should be reflected there.
My dream is to see G5.1 and G5.2 one day for Quadratic/NURBS Spline ;)

terjeio, 2022-07-13:
I'll add G5.1 soon, it is a matter of transforming the single control point into two for a cubic spline. Do you have any test files complete with a plot of the expected parh that I can use for verification?

NURBS is more complicated as a dynamic array of control points is needed - and some of the processors supported by grblHAL do not have much heap available for that.
BTW are there any CAM tools outputting NURBS paths that warrants the work involved in making NURBS available?

terjeio, 2022-07-13:
I could not find any G5.1 examples via the links you provided.

> I guess it's not that hard to extract NURBS from Rhino objects.

I am not able to do that, are you?

BTW the [latest edge version of ioSender](http://www.io-engineering.com/downloads) should now render G5.1 paths correcly, and has initial support for rendering multi turn arcs (G2/G3 with P parameter). Note that grblHAL is not yet ready to handle these.

kimstik, 2022-07-28:
one more G5 sample.
imported from https://upload.wikimedia.org/wikipedia/commons/f/ff/GClef.svg
juicy-gcode.exe GClef.svg -b -o GClef.nc
[GClef.zip](https://github.com/grblHAL/core/files/9206676/GClef.zip)