grblHAL/core #156: hi,terjeio,Command "M3" without S parameter, the pin cannot be set to high level, is it illegal?

hankecnccc, 2022-06-30:
hi, terjeio, I'm glad to find a cheap and easily available chip "STM32F401RCT6" in this age of chip shortage, I used it to make a GRBLHAL board for sale, I found in my tests that only send when powered "M3", without the "S" parameter value, will not make the "spindle start" pin set to a high level, that is, the spindle starts, because the default "S" value is 0 when powered on, but is this illegal? The "M3" command is only the spindle turn-on signal, and should not be affected by the "S" value. The spindle PWM will be affected by the "S" value.
Finally, I wish you a happy life! !

hankecnccc, 2022-07-01:
While waiting for your reply, I saw the explanation of parameter 7

closed 2022-07-01 by hankecnccc

terjeio, 2022-07-04:
Fixed in [version 20220703](https://github.com/grblHAL/core/blob/master/changelog.md).

hankecnccc, 2022-07-04:
So, is the other question I reported also a bug?

hankecnccc, 2022-07-04:
> Fixed in [version 20220703](https://github.com/grblHAL/core/blob/master/changelog.md).

You seem to be busy lately? Also, good news, there are many people in China who are interested in your grblhal and they have pre-ordered the grblhal stm32F4 board that I made