grblHAL/core #158: Trinamic support question

dresco, 2022-07-04:
Hi @terjeio 

I would like to get Trinamic support into the H7 port (for the BTT SKR v3 board). I've bought a board with 2209 drivers for testing, and @Demitrius has a board with 5160 drivers on the way. Looking at the board pinouts, will need to use both soft UART and soft SPI.

Am looking at the F4 code as a starting point, and something puzzles me. The following code is not reached if built for 2209, as is inside an earlier `#if TRINAMIC_SPI_ENABLE` directive.. Is this a bug, or left over from the recent refactoring?

https://github.com/grblHAL/STM32F4xx/blob/be9965052f8ec7d0e56ccf56ca7c3c06123981cd/Src/btt_skr_2.0.c#L192-L195

terjeio, 2022-07-04:
It is left over from refactoring. For soft UART [tmc_uart.c](https://github.com/grblHAL/STM32F4xx/blob/master/Src/tmc_uart.c) is used. I'll remove the stray code in the next commit.

dresco, 2022-07-09:
I've added TMC2209 support with soft UART, and appears to be working okay on my BTT SKR 3 board with 2209 drivers. Have pushed these updates to my [H7 repo](https://github.com/dresco/STM32H7xx).

However, the TMC5160 is going to need a software SPI implementation for this BTT board, @terjeio I don't suppose you have anything like this already?

terjeio, 2022-07-09:
Great. I do not have any soft SPI code at hand but it should be easier to implement than soft UART as the clock is supplied by the master when receiving data.

dresco, 2022-07-10:
Cool, I'll keep an eye out for the TMC5160-BOB being back in stock anywhere & pick one up for testing..

dresco, 2022-08-05:
Trinamic support for 2209 (soft UART) and 5160 (soft and hard SPI) is now in the H7 master branch.

closed 2022-08-05 by dresco