grblHAL/core #159: $37 - Stepper deenergize mask  dont work

gendebeat, 2022-07-06:
Hello im try to disable deenergize in some axis, but only works in all 3 axis, is the mask work correct?  I use iMXRT1062 with T41U5XBB board. and enable pin is NOT the same in all axis. 

Previously i use smoothieware firmware and i have a M18 M17  code to disable and enable steppers (like marlin).Because i like 
to fast move my belt machine by hand. If all steppers are energized ,how i cant enable/disable it ? (previosly i use cncjs with a button with Mcode) 
thanks

terjeio, 2022-07-08:
Changing the $37 setting will immediately change the stepper enable outputs for the iMXRT1062 driver. Note that the setting is for which enable outputs _not_ to disable.

M17/M18 support can be added via a small plugin. Since M17/M18 are not standard gcodes I do not want to add these to the core.

terjeio, 2022-07-08:
I've added a [plugin for  M17/M18 (and M84)](https://github.com/grblHAL/Templates/tree/master/my_plugin/Stepper%20enable%20control).

gendebeat, 2022-09-06:
HI, 
until today I could not try the pluggin in my machine, but he's doing weird things.
When i manual move, the machine act late. like some delays receiving commands. I cant move in one axis and immediately on the other. The same if i do a diagonal movement. I dont try sending a gcode, but....
I compile without the plugin, it work correctly. I use CNCJS as interface for working.

In addition, i try this M codes and still do the same that using $ configuration. 
Cant disable axis by axis. Is all or nothing.
In the T41U5XBB board the enable pins are independent. Maybe a problem with iMXRT1062 code?
thanks

gendebeat, 2022-09-06:
ok, 
i disable #define MPG_ENABLE and now the movement are in realtime with the custom plugin.
I had enabled MPG, because i want in future use a pendant.

The issue with disable one axis at time, continues

terjeio, 2022-09-06:
> Cant disable axis by axis. Is all or nothing.

Do you have command examples that fails? And can you debug the plugin code, it has been a little while since I wrote it for you...

gendebeat, 2022-09-06:
sorry, was a mistake in my hardware. The T41U5XBB board have the enable out for axis jumpered with X.

closed 2023-04-29 by terjeio