grblHAL/core #166: How does Cycle start & cycle hold signals work?

muhammadelmogy, 2022-07-27:
I want to know how does Cycle start & cycle hold signals work

I see in some designs (Minimal Black Pill & grblHAL-teensy-4.x) they have separate input for each function
Ok for start, what will happen when I press it, what exactly it will start, the Gcode is located on the PC(ioSender or bCNC)
will the controller board send a signal to the PC and ask it to start sending G-code lines?

For the cycle hold will this do the same as start but for pausing program execution?

Can start and hold use the same input? when press first time start and when press a second time hold the program flow?

terjeio, 2022-07-28:
> Can start and hold use the same input? 

Yes, if you write plugin or board support code for it. You can redirect the [control signals  interrupt callback](https://github.com/grblHAL/core/blob/26fa15a9900213d790d7a600541b863d35f0e8f6/hal.h#L190) to your code and change the hold pin signal to the cycle start pin signal either as a toggle or depending on the state.

muhammadelmogy, 2022-07-28:
ok thanks, about the other questions:

> what will happen when I press it, what exactly it will start, the Gcode is located on the PC(ioSender or bCNC) will the controller board send a signal to the PC and ask it to start sending G-code lines?
> 
> For the cycle hold will this do the same as start but for pausing program execution?

terjeio, 2022-07-28:
If a program is loaded in ioSender pressing cycle start will start the program, if a hold is active or a tool change is pending it will resume running the program. I do not know how bCNC handles cycle start if a program is loaded and cycle start is pressed when a hold is not active.

Hold will pause the program until cycle start is pressed.

closed 2022-07-28 by muhammadelmogy