grblHAL/core #167: hi, is terjeio there? grblhal if only one system parameter is set at a time, then this system parameter will not be saved after power failure

hankecnccc, 2022-07-28:
hi, is terjeio there? grblhal if only one system parameter is set at a time, then this system parameter will not be saved after power failure

I need to set more than 2 parameters each time, the parameter will be saved

hankecnccc, 2022-07-28:
On F401CCU6 and F401RCT6, I don't know about other microcontrollers

terjeio, 2022-07-30:
I do not see this issue with a F446 which shares the same code. Do you change the parameter that fail to stick with the `$` settings command directly or via a sender settings UI when this happens?

hankecnccc, 2022-07-30:
> I do not see this issue with a F446 which shares the same code. Do you change the parameter that fail to stick with the `$` settings command directly or via a sender settings UI when this happens?

Thank you for your reply, I manually sent the command "$x=x", this bug can be reproduced accurately, it is known that in F401RCT6 and F401CCU6, "F446" may not be a problem, can you move this problem to STM32F4xx? ? Maybe someone will fix it

hankecnccc, 2022-07-30:
> I do not see this issue with a F446 which shares the same code. Do you change the parameter that fail to stick with the `$` settings command directly or via a sender settings UI when this happens?
[image]

This is the control board I made based on grblhal

terjeio, 2022-07-30:
> I manually sent the command "$x=x", this bug can be reproduced accurately

I do not have a F401 to check with - the code difference between F446 and F401 is only the clock setup and the loader script since the flash memory layout is different.

> can you move this problem to STM32F4xx? ? Maybe someone will fix it

Yes, that someone has to be someone with access to a F401 and preferably a debugger. The initial place to [set a breakpoint is here](https://github.com/grblHAL/core/blob/7dac639da81f4518bb2890d17b84a06867b96b8a/nvs_buffer.c#L310). This is where most of the settings are written on a change.

Nice board BTW and I guess you are storing the settings in flash since I cannot see any EEPROM/FRAM chip.