grblHAL/core #168: How often an Aux input can be used?

muhammadelmogy, 2022-07-28:
I see in the code there is support for Aux Inputs, I want to know what would be the case where I would need an Aux input in the controller.
For Aux outputs, maybe I someone will use them to turn on extra stuff using some standard G-code but for Aux input what can I use them for?

terjeio, 2022-07-28:
User plugin code, or "standard" plugins such as the one for [Bluetooth comms](https://github.com/grblHAL/Plugins_Bluetooth).

muhammadelmogy, 2022-07-28:
But there is not major use in the industry for example read some sensors (I am not really expert with CNC machines)
Do the need to be interrupt supporting?
I want to eliminate any unnecessary pins on the board I am designing so I can have them for other valuable functions !

terjeio, 2022-07-28:
Unneccesary pins for one user is vital for another... To make the perfect controller you will need a lot of pins.

The Blueetooth plugin requires an interrupt capable pin as well as an UART port. The plasma plugin needs an analog input and arc ok input etc etc...

muhammadelmogy, 2022-07-28:
> Unneccesary pins for one user is vital for another... To make the perfect controller you will need a lot of pins.

You are right, no doubts about it, but I am more concerned about CNC routers and Also I am still trying to make use of what I have on hand right now which is stm23f411 chip

5ocworkshop, 2022-08-13:
> 

> I see in the code there is support for Aux Inputs, I want to know what would be the case where I would need an Aux input in the controller.
> For Aux outputs, maybe someone will use them to turn on extra stuff using some standard G-code but for Aux input what can I use them for?

As an example, I use one of the Aux Inputs to connect a simple momentary switch that allows me to over ride my status lights and switch my RGB gantry lights to "white" so I can better inspect my work.  So I have an "inspection light" button on my button box for this that uses the Aux Input.

5ocworkshop, 2022-08-13:
Another, perhaps more common example, is the "enclosure door".  Where you want the machine to respond to the safety door to the enclosure being opened while it is running.