grblHAL/core #174: use second uart

MARIOBASZ, 2022-08-16:
I would like to be able to send current position data to a second microprocessor, (I don't know whether to receive information). Is it easy to do?

terjeio, 2022-08-16:
Which driver?

MARIOBASZ, 2022-08-16:
I am using Arduino Due.

terjeio, 2022-08-16:
Ok, then the easiest is perhaps enabling MPG_MODE and parsing the data coming from the MPG port. If you are connecting via the programming port another option is to listen in on the TX pin. Both these methods requires a sender requesting a status report at regular intervals or a plugin triggering it when no sender is connected.

If you want your own format and automatic pushing of position data you can write [a plugin](https://github.com/grblHAL/Templates/tree/master/my_plugin) that [opens an UART stream](http://svn.io-engineering.com/grblHAL/html/stream_8h.html#a5fd022599405a1221dceac5559b96587) and outputs the data you need. You can use the tick timer (1ms interval) in combination with subscribing to the `on_execute_realtime` event. An example of how to produce a regular event is in [this plugin](https://github.com/grblHAL/Templates/tree/master/my_plugin/stm32F411blinky).
I do not have any straigth example code for how to open an UART - the closest ones are the [Bluetooth](https://github.com/grblHAL/Plugins_Bluetooth/blob/230a01864a5d8376f022888544ba05ac91475f90/hc_05.c#L312-L324) or [ModBus](https://github.com/grblHAL/Plugins_spindle/blob/a3450d236c09818e8fef1c25b9385683993f00a0/modbus.c#L497-L552) plugins.

MARIOBASZ, 2022-08-16:
Is it okay to listen to Tx while connected to the computer? If it doesn't cause any inconvenience, I would start by trying that way.

terjeio, 2022-08-17:
> Is it okay to listen to Tx while connected to the computer?

Yes, I have done so with other boards.

MARIOBASZ, 2022-08-21:
Thanks. I'll try that way.

closed 2023-04-29 by terjeio